gerbv/gerbv #15: ./run_test fails on Unix

eyal0, 2021-09-03:
To reproduce, make and install the code and then:

```
cd test
./run_tests.sh
```

I'm seeing this result:

```
Passed 67, failed 23, skipped 0 out of 90 tests.
```

I'll start with a git bisect to see where it was introduced.

eyal0, 2021-09-03:
I went back a few years and as far as I can tell, these tests have been broken for a long, long time.  Looks like maybe no one touched them since 2008!

They are golden file tests: gerbv makes an output in the form of an image and the image is compared against the expected image to see if there are any changes.

Seeing as the tests have been failing for so long, rather than try to figure out what's going on with each one, it could be a better strategy to just regenerate all the golden files so that the tests are in working order again.  And then we can add the testing into CI and have it for detecting future bugs.

ooxi, 2021-09-03:
I tend to agree @eyal0. I think it's more important to get a new test suite up and running then trying to get the old tests working again.

Nevertheless I am a bit torn about the old test design, since it is comparing images to well known golden samples. Is this portable at all with regards to hardware acceleration for graphic operations?

Using golden images is however good for manual verification.

eyal0, 2021-09-03:
> Nevertheless I am a bit torn about the old test design, since it is comparing images to well known golden samples. Is this portable at all with regards to hardware acceleration for graphic operations?

Good question.  I have seen errors in my own code with similar things.  For example, we might find that the precision of `float` or `double` is different on different platforms and the tests fail on some computers.  Or optimizations by different compilers that introduce minor floating point errors.  Or underlying libraries like the gtk that behave differently.  I have dealt with stuff like this before.

But having testing in CI is too good to pass up!  My suggestions is that we keep it for now and, should it prove to be a hassle in the future, then we fix it.  Only if we see it really misbehaving do we need to do something about it.

ooxi, 2021-09-04:
> But having testing in CI is too good to pass up! My suggestions is that we keep it for now and, should it prove to be a hassle in the future, then we fix it. Only if we see it really misbehaving do we need to do something about it.

That sounds good!

PR #18 (2021-09-04): test: Regenerate all golden files
The unit testing has not been working for a long time.  This
regnerates the golden files so that we can start fresh.

This fixes #15
closed 2021-09-04 by eyal0