gerbv/gerbv #26: Unify header include guards

ooxi, 2021-09-23:
libgerbv currently uses multiple different styles of include guards:

* Invalid: Include guards like `__COMMON_H__` which collide with reserved identifiers (7.1.3)
* Name-Clash: Short include guard defines like `CSV_H` which might collide with other projects, since they don't use a project prefix
* Missing: No include guard present, which might lead to redefinitions
* Prefixed: Include guards like `GERBV_LAYERTYPE_PICKANDPLACE_H` which are both valid identifiers and do not risk collission with other projects
* Vendored: Header copied from or generated by third party

I propose to add prefixed include guards (like `GERBV_CSV_H`) to all headers unless they are vendored

## Pros

* No risk of include guard collissions due to project specific prefix
* No risk of redefinitions due to missing include guards
* No risk of unexpected behaviour since we are not using reserved identifiers

## Cons

* We change the defined symbols, which might break source compatibility for very fragile users

## Alternatives

We could use `#pragma once` instead, which is supported by all platforms we care about. Nevertheless I would refrain from using a non standard feature.

## Overview

| Header                                     | Include guard                    | Category   |
| ------------------------------------------ | -------------------------------- | ---------- |
| [`amacro.h`](https://github.com/gerbv/gerbv/blob/main/src/amacro.h)                 | `AMACRO_H`                       | Name-Clash |
| [`attribute.h`](https://github.com/gerbv/gerbv/blob/main/src/attribute.h)           | `ATTRIBUTE_H`                    | Name-Clash |
| [`callbacks.h`](https://github.com/gerbv/gerbv/blob/main/src/callbacks.h)           | Missing                          | Missing    |
| [`common.h`](https://github.com/gerbv/gerbv/blob/main/src/common.h)                 | `__COMMON_H__`                   | Invalid    |
| [`csv_defines.h`](https://github.com/gerbv/gerbv/blob/main/src/csv_defines.h)       | `DEFINES_H`                      | Name-Clash |
| [`csv.h`](https://github.com/gerbv/gerbv/blob/main/src/csv.h)                       | `CSV_H`                          | Name-Clash |
| [`draw-gdk.h`](https://github.com/gerbv/gerbv/blob/main/src/draw-gdk.h)             | `DRAW_GDK_H`                     | Name-Clash |
| [`draw.h`](https://github.com/gerbv/gerbv/blob/main/src/draw.h)                     | `DRAW_H`                         | Name-Clash |
| [`drill.h`](https://github.com/gerbv/gerbv/blob/main/src/drill.h)                   | `DRILL_H`                        | Name-Clash |
| [`drill_stats.h`](https://github.com/gerbv/gerbv/blob/main/src/drill_stats.h)       | `DRILL_STATS_H`                  | Name-Clash |
| [`dynload.h`](https://github.com/gerbv/gerbv/blob/main/src/dynload.h)               | `DYNLOAD_H`                      | Name-Clash |
| [`gerber.h`](https://github.com/gerbv/gerbv/blob/main/src/gerber.h)                 | `GERBER_H`                       | Name-Clash |
| [`gerb_file.h`](https://github.com/gerbv/gerbv/blob/main/src/gerb_file.h)           | `GERB_FILE_H`                    | Name-Clash |
| [`gerb_image.h`](https://github.com/gerbv/gerbv/blob/main/src/gerb_image.h)         | `GERB_IMAGE_H`                   | Name-Clash |
| [`gerb_stats.h`](https://github.com/gerbv/gerbv/blob/main/src/gerb_stats.h)         | `gerb_stats_H`                   | Name-Clash |
| [`gerbv.h`](https://github.com/gerbv/gerbv/blob/main/src/gerbv.h)                   | `__GERBV_H__`                    | Invalid    |
| [`gerbv_icon.h`](https://github.com/gerbv/gerbv/blob/main/src/gerbv_icon.h)         | Missing                          | Vendored   |
| [`gettext.h`](https://github.com/gerbv/gerbv/blob/main/src/gettext.h)               | `_LIBGETTEXT_H`                  | Vendored   |
| [`icons.h`](https://github.com/gerbv/gerbv/blob/main/src/icons.h)                   | Missing                          | Vendored   |
| [`interface.h`](https://github.com/gerbv/gerbv/blob/main/src/interface.h)           | Missing                          | Missing    |
| [`lrealpath.h`](https://github.com/gerbv/gerbv/blob/main/src/lrealpath.h)           | `__LREALPATH_H__`                | Invalid    |
| [`main.h`](https://github.com/gerbv/gerbv/blob/main/src/main.h)                     | `MAIN_H`                         | Name-Clash |
| [`opdefines.h`](https://github.com/gerbv/gerbv/blob/main/src/opdefines.h)           | Missing                          | Missing    |
| [`pick-and-place.h`](https://github.com/gerbv/gerbv/blob/main/src/pick-and-place.h) | `GERBV_LAYERTYPE_PICKANDPLACE_H` | Prefixed   |
| [`project.h`](https://github.com/gerbv/gerbv/blob/main/src/project.h)               | `PROJECT_H`                      | Name-Clash |
| [`render.h`](https://github.com/gerbv/gerbv/blob/main/src/render.h)                 | Missing                          | Missing    |
| [`scheme.h`](https://github.com/gerbv/gerbv/blob/main/src/scheme.h)                 | `_SCHEME_H`                      | Invalid    |
| [`scheme-private.h`](https://github.com/gerbv/gerbv/blob/main/src/scheme-private.h) | `_SCHEME_PRIVATE_H`              | Invalid    |
| [`selection.h`](https://github.com/gerbv/gerbv/blob/main/src/selection.h)           | Missing                          | Missing    |
| [`table.h`](https://github.com/gerbv/gerbv/blob/main/src/table.h)                   | `TABLE_H`                        | Name-Clash |

ooxi, 2021-09-23:
@efa / @eyal0 / @spe-ciellt: While this is borderline bike shedding, I would love to hear your input since I'm hesitant to perform style changes without community support.

eyal0, 2021-09-23:
I usually use pragma once because it's easy and at some point we have to ditch the past. But yeah, there's always that one programmer on GCC 2.0 or something that will yell. Whatever you prefer. I'd go with pragma once and switch to something else only if someone complains in the future.

ooxi, 2021-09-23:
If nobody disagrees I will go ahead with your suggestion @eyal0

spe-ciellt, 2021-09-23:
Both `static_assert()` and `#pragma once` is C++ constructs. I have nothing against C++, I would love to use C++ in gerbv. But just as an information. Now I noticed that they are somewhat available in C as well:
- https://en.cppreference.com/w/c/preprocessor/impl
- https://en.cppreference.com/w/c/error/static_assert

efa, 2021-09-23:
I'm in favour of C standard suggestion:
```
#ifndef LIBRARY_FILENAME_H
#define LIBRARY_FILENAME_H
// contents of the header
#endif /* LIBRARY_FILENAME_H */
```

eyal0, 2021-09-23:
gcc has pragma once since 2003.  https://gcc.gnu.org/gcc-3.4/changes.html

clang has had it for 14 years: https://github.com/llvm-mirror/clang/blame/aa231e4be75ac4759c236b755c57876f76e3cf05/lib/Lex/Pragma.cpp#L394

Maybe we'll use pragma once and it will be the last straw, getting some user out there on a 20-year-old compiler to finally upgrade.  We're doing the world a favor.  :-)

efa, 2021-09-23:
I saw in the documentation is writted that a supported platform is:
Solaris (5.7 and 5.8)
are we sure che Sun compiler support pragma?

ooxi, 2021-09-30:
| User | Opinion |
| --- | --- |
| @efa | Pro `#ifndef` |
| @eyal0 | Pro `#pragma` |
| @ooxi | Pro `#ifndef` |
| @spe-ciellt | Indifferent / Slight reservations against `#pragma` |

I therefore suggest going ahead with using `#ifndef`. Since we don't have too many headers, the maintenance burden will be more or less zero.

efa, 2021-09-30:
I do not see a PR for this, it is not integrated yet. You should keep open this till PR closure

efa, 2021-10-01:
any volunteer? If not I propose myself, it is easy, should I have time in the week end

ooxi, 2021-10-01:
> any volunteer? If not I propose myself, it is easy, should I have time in the week end

@efa I would love if you could help me out with this PR, since I cannot get onto this before next week :-)

efa, 2021-10-03:
done in https://github.com/gerbv/gerbv/pull/29

efa, 2021-10-04:
unfortunately the check do not passed. Looking in the log I saw two tests failed, one for different output and the other for a segmentation fault. I tryed to run the checks manually and fail.
So I will re-clone the original repo, and .h file by .h file, I will try wich one generate the trouble. This is not a fast task, as the .h files are 30 files

efa, 2021-10-04:
without `opdefines.h` modified I have no segmentation fault

efa, 2021-10-04:
about the other test that fail for different output, I got a FAIL also with current (umodified) git source:

```
Test:  example_pick_and_place_LED
../src/run_gerbv -- --export=png --window=640x480 --output=outputs/example_pick_and_place_LED.png  ./inputs/../../example/pick-and-place/LED.xy
Set LD_LIBRARY_PATH to /home/efa/c/gerbv-2021-10-03github/src/.libs:/home/efa/lib::/usr/lib:/usr/local/lib
Set LD_RUN_PATH to /home/efa/c/gerbv-2021-10-03github/src/.libs
Set LD_PRELOAD to /home/efa/c/gerbv-2021-10-03github/src/.libs/libgerbv.so
FAILED:  See mismatch/example_pick_and_place_LED
```

running:
gerbv-2021-10-04github/test$ mismatch/example_pick_and_place_LED/animate.sh

I cannot find visual differences, but if it fail, the output is not bit-exact (also with current sources). I do not know how can pass CI with current sources

efa, 2021-10-04:
will do a new PR without patching `opdefines.h`, than will debug what happen with gdb

efa, 2021-10-04:
finally CI passed

ooxi, 2021-11-19:
Oh man I totally missed the progress on this PR, sorry about that :-(

Will review and merge in a bit!

rampageservices, 2026-03-02:
Hi all — I was reviewing the state of this issue and wanted to summarize where things stand, since it's been a while.

**What's been done:**
- PR #141 by @ooxi was merged, fixing the include guard in `draw.h`
- That's the only file that was addressed

**What wasn't merged:**
- PR #29 by @efa — closed due to a segfault caused by the `opdefines.h` change, plus a CI failure
- PR #142 by @ooxi — closed after the CI coverage build failed

**What still needs fixing (current `main`):**
- 5 headers with reserved `__DOUBLE_UNDERSCORE__` identifiers (`__COMMON_H__`, `__GERBV_H__`, etc.)
- 14 headers with short guards lacking a `GERBV_` prefix (`CSV_H`, `DRILL_H`, `MAIN_H`, etc.)
- 2 headers with no include guard at all (`render.h`, `selection.h`)
- Only `pick-and-place.h` uses the agreed-upon `GERBV_` prefixed pattern

@ooxi @efa — would either of you like to reopen your PRs and finish this out? If not, I'm happy to put together a single PR that addresses all remaining headers, using the `GERBV_FILENAME_H` convention that was agreed on. I'd include co-author attribution for both of you given the prior work and discussion.