gerbv/gerbv #27: No error check for invalid arcs in single quadrant

miltonneal54, 2021-09-30:
File: Gerber.c
Method: calc_cirseg_sq()
This method does not check for invalid arcs.
As stated in the RS274X specification:
Single quadrant circular interpolation plots an arc within one quadrant (90°).
Single quadrant arcs must fit entirely within the quadrant in which they begin. A
separate data block is required for each quadrant. A minimum of four data blocks
is required to plot a circle.
This is evident in the the file "test-circular-interpolation-1.gbx.

G04 Switch to quadrant mode, draw clockwise*
G04 Note: since this is single quadrant mode, I and J must be*
G04  positive, and the parser should automatically negate the J value*
G04  to make the curve travel in the clockwise direction*
G74*
G01X00400Y**00300**D02*
G02X00500Y**00300**I00150J00300D01*

In the last arc code, StartY and StopY are the same value, (3) meaning they are in different quadrants.
Rather than catching this error it renders an incorrect arc segment. This can be seen in the output (the top right arc) that the StartY and StopY are clearly not the same. Admitting, this is only a hand coded sample, and may not happen in a practical file, but I think this should be checked anyway.
I also noted that the circle segment parameters store the arc segment's  width and height, which are mostly different values. As these are used to render the arc segment the result is a distorted arc (see the lower right arc in the same file). Perhaps the width and height should store the diameter as in the MQ arc segment code.
After all this, the G74 code has been deprecated in the X3 specification so is only necessary to fix if you wish to support legacy pcb gerbers.
With thank Milton.

ooxi, 2021-09-30:
Thank you very much for catching this! It would be great if we could work together on fixing this issue. Would you like to propose a patch?

eyal0, 2021-10-01:
Arcs are hard because they are over specified. An arc is always a part of a circle. A circle can be defined by a center point and a radius. And the arc is described by a start angle and end angle. So that's a total of 5 numbers: centerx centery radius angle0 angle1.

Gcode uses startx, starty endx endy i j. Six numbers. It's over specified. So it's possible to make illegal arcs.

What to do about it? First of all, some arcs will always appear illegal due to rounding. So you want to allow for a little bit of error in the calculation.

It's really important that the endpoint is maintained. So if you need to fudge a number, it better be the radius location.

Also, say it's illegal because the numbers are off by more than a rounding error. What do you want to do about it? Crash? Emit an error? Emit a warning? Silently interpolate a good answer? For pcb2gcode, we interpolate. The start is right, the end is right, and the middle parts might get nudged around a bit such that the result might be a part of a spiral instead of a part of a circle.

There's the programming adage: be strict in what you output and liberal in what you accept. So is gerbv more useful when it's really strict about arcs or is it more useful when it's able to process some malformed arcs?

I'd vote for the spiral solution because it maintains start and end and it can still handle all inputs. Maybe emit a warning if it's too spirally. Like if the start and end radii differ by more than some percentage.

ooxi, 2021-10-01:
> I'd vote for the spiral solution because it maintains start and end and it can still handle all inputs. Maybe emit a warning if it's too spirally. Like if the start and end radii differ by more than some percentage.

That sounds like a quick win! @miltonneal54 would you be willing to help us out with such a PR?

miltonneal54, 2021-10-04:
My interest in the Gerbv project is mostly in the libgerbv library which I'm "porting" to a .Net C# library. During some testing on my project is where I came across this issue of malformed sq arcs.
 I agree with @eyal0  and @ooxi  resolution and once we come up with a good percentage margin should not be difficult to implement. The online Ucamco Gerber Viewer seems to treat "arc deviation" this way.
The other function I wish to implement is interpolating sq arc segment across two adjacent quadrants. The same quadrant rule seems to be dropped in the X2 Format Specification (Revision 2015.10). Again the Ucamco online app parses and render these type of arcs without complaint.
Gerbv (and naturally my library) currently don't support this type of sq arc. I'm currently working on the code to achieve this. (C and C#).
I'm relatively new to Github and never generated a PR, I'll see how we go :)

ooxi, 2021-11-19:
@miltonneal54 I'm closing this issue since there are no actionable items for us. Feel free to create a pull request whenever you are ready :-)

closed 2021-11-19 by ooxi