gerbv/gerbv #45: Remove all compile-time warnings

eyal0, 2021-11-01:
There are a few.  To see them, run:

```
./configure
```

Then go into each `Makefile` and modify the compilation so that all warnings are treated as errors.  You can search the Makefile for `-O` and add `-Werror` on to the line.  Then run `make`.  The compile will fail.

Here are two errors:

```
interface.c: In function ‘interface_create_gui’:
interface.c:340:2: error: ‘gdk_pixbuf_new_from_inline’ is deprecated [-Werror=deprecated-declarations]
  pointerpixbuf = gdk_pixbuf_new_from_inline(-1, pointer, FALSE, NULL);
  ^~~~~~~~~~~~~
In file included from /usr/include/gdk-pixbuf-2.0/gdk-pixbuf/gdk-pixbuf.h:34,
                 from /usr/include/gtk-2.0/gdk/gdkpixbuf.h:37,
                 from /usr/include/gtk-2.0/gdk/gdkcairo.h:28,
                 from /usr/include/gtk-2.0/gdk/gdk.h:33,
                 from /usr/include/gtk-2.0/gtk/gtk.h:32,
                 from gerbv.h:73,
                 from interface.c:29:
/usr/include/gdk-pixbuf-2.0/gdk-pixbuf/gdk-pixbuf-core.h:362:12: note: declared here
 GdkPixbuf* gdk_pixbuf_new_from_inline (gint          data_length,
```

```
gerb_file.c: In function ‘gerb_find_file’:
gerb_file.c:323:4: error: ‘strncat’ output truncated before terminating nul copying as many bytes from a string as its length [-Werror=stringop-truncation]
    strncat(complete_path, filename, strlen(filename));
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

ooxi, 2021-11-02:
Getting rid of all those warnings will be a major step forward in source code hygiene :+1:

ooxi, 2021-11-04:
@eyal0 all GCC compile time warnings are fixed with d7f42a9afa62b3b743a4cb35050fa74c2a17c412. How about renaming this ticket to _Remove all compile-time warnings in GCC_ and closing it, since Clang warnings will be tracked in issue #60?