gerbv/gerbv #46: Remove deprecated gdk_pixbuf_new_from_inline

eyal0, 2021-11-01:
This is one of the warnings when compiling.

This issue is a subtask of #45 

```
interface.c: In function ‘interface_create_gui’:
interface.c:340:2: error: ‘gdk_pixbuf_new_from_inline’ is deprecated [-Werror=deprecated-declarations]
  pointerpixbuf = gdk_pixbuf_new_from_inline(-1, pointer, FALSE, NULL);
  ^~~~~~~~~~~~~
In file included from /usr/include/gdk-pixbuf-2.0/gdk-pixbuf/gdk-pixbuf.h:34,
                 from /usr/include/gtk-2.0/gdk/gdkpixbuf.h:37,
                 from /usr/include/gtk-2.0/gdk/gdkcairo.h:28,
                 from /usr/include/gtk-2.0/gdk/gdk.h:33,
                 from /usr/include/gtk-2.0/gtk/gtk.h:32,
                 from gerbv.h:73,
                 from interface.c:29:
/usr/include/gdk-pixbuf-2.0/gdk-pixbuf/gdk-pixbuf-core.h:362:12: note: declared here
 GdkPixbuf* gdk_pixbuf_new_from_inline (gint          data_length,
```

eyal0, 2021-11-01:
A GDK pixbuf is a an internal format for GDK to store an image.  It's a little bit like an XPM file.  There are four of them in the icons.h file.  I did not find the original images but I will look for them.

If we can find the original XPM, we can convert the pixbuf into XPM data, the loading of which is not deprecated.

Alternatively, we can convert the images from pixbuf into xpm but then we'd need to figure out how to do that.  Tools might exist for it though the only tool that I have found is `gdk-pixbuf-csource` but it is doing the reverse transformation.

The images are four, for move, zoom, and measure.  I think that they might be icons for the program or maybe mouse pointers?

eyal0, 2021-11-01:
Looks like they were added in 82579dfd66950e892fc1a1325714b4c91eac3d16 but we don't have the original drawings in the source code, which is a bummer.

eyal0, 2021-11-01:
https://github.com/krh/gtk/blob/master/gdk-pixbuf/gdk-pixdata.h#L53-L66

This comment seems to explain the format of the data.  Maybe we can just use this function instead and maintain the same data:

https://developer.gimp.org/api/2.0/gdk-pixbuf/gdk-pixbuf-creating.html#gdk-pixbuf-new-from-data

According to this file `/usr/include/gdk-pixbuf-2.0/gdk-pixbuf/gdk-pixbuf-core.h`, the function is not deprecated, which would remove the warning, though in a kind of crappy way.  But maybe good enough for now, until the next deprecation!

XPM would be nicer and GResource would be even better but those would be more complicated.

eyal0, 2021-11-01:
from data seems to be working okay.  For testing, I did this:

```diff
diff --git a/src/interface.c b/src/interface.c
index feeef13..db5082a 100644
--- a/src/interface.c
+++ b/src/interface.c
@@ -337,7 +337,13 @@ interface_create_gui (int req_width, int req_height)
 		screen.settings = g_settings_new(settings_id);
 	}
 
-	pointerpixbuf = gdk_pixbuf_new_from_inline(-1, pointer, FALSE, NULL);
+	pointerpixbuf = gdk_pixbuf_new_from_data(pointer+24,
+                                                 GDK_COLORSPACE_RGB,
+                                                 1,
+                                                 8,
+                                                 24,
+                                                 24,
+                                                 96,NULL,NULL);
 	movepixbuf = gdk_pixbuf_new_from_inline(-1, move, FALSE, NULL);
 	zoompixbuf = gdk_pixbuf_new_from_inline(-1, lzoom, FALSE, NULL);
 	measurepixbuf = gdk_pixbuf_new_from_inline(-1, ruler, FALSE, NULL);
```

Here's what gerbv looks like in the header bar now:

[image]

Looks good.  Same as the original.  Just to see that my changed have made a difference, here's what happens if I don't `+24`:

[image]

See how the pointer is now shifted over to the right?

Here it is magnified with xmag:

[image]

And again zoomed on the garbage in the corner:

[image]

That's 6 pixels of garbage and everything is slid over by 6 pixels.  Those 6 correspond to the 24 bytes, 4 bytes per pixel, which is the RGB channels plus another byte for alpha, so RGBA.  That's why it's 6.

eyal0, 2021-11-01:
Here's a kludge to print convert the run-length encoded data to not rle.  I think that it works:

```diff
diff --git a/src/interface.c b/src/interface.c
index ebe8bfc..a0d9e76 100644
--- a/src/interface.c
+++ b/src/interface.c
@@ -45,8 +45,10 @@
 #include "interface.h"
 #include "render.h"
 #include "selection.h"
+#include <gdk-pixbuf/gdk-pixdata.h>
 
 #include "draw.h"
+#include <ctype.h>
 
 #include "gerbv_icon.h"
 #include "icons.h"
@@ -354,6 +356,43 @@ interface_create_gui (int req_width, int req_height)
 	movepixbuf = pixbuf_from_icon(&move);
 	zoompixbuf = pixbuf_from_icon(&lzoom);
 	measurepixbuf = gdk_pixbuf_new_from_inline(-1, ruler, FALSE, NULL);
+        GdkPixdata foo2;
+        gdk_pixdata_from_pixbuf(&foo2, measurepixbuf, FALSE);
+        guchar* foo = foo2.pixel_data;
+        int width_so_far = 0;
+        printf("hello\n");
+        for (int i = 0; i < 96*24; i++) {
+          if (width_so_far == 0) {
+            fprintf(stderr, "  \"");
+            width_so_far += 3;
+          }
+          // from https://github.com/GNOME/gdk-pixbuf/blob/19512e34675b658e7509957b4f99bff2c3d56979/gdk-pixbuf/gdk-pixdata.c#L628
+          if (foo[i] < 33 || foo[i] > 126 || foo[i] == '?') {
+            fprintf(stderr, "\\%o", foo[i]);
+            width_so_far++;
+            if (foo[i] < 010) {
+              width_so_far++;
+            } else if (foo[i] < 0100) {
+              width_so_far+=2;
+            } else {
+              width_so_far+=3;
+            }
+          } else if (foo[i] == '\\') {
+            fprintf(stderr, "\\\\");
+            width_so_far+=2;
+          } else if (foo[i] == '"') {
+            fprintf(stderr, "\\\"");
+            width_so_far+=2;
+          } else {
+            fprintf(stderr, "%c", foo[i]);
+            width_so_far++;
+          }
+
+          if (width_so_far >= 72) {
+            fprintf(stderr, "\"\n");
+            width_so_far = 0;
+          }
+        }
 
 	tooltips = gtk_tooltips_new();
 	accel_group = gtk_accel_group_new ();
```

eyal0, 2021-11-01:
[image]

New above, old below.  Looks pretty good to me.

ooxi, 2021-11-02:
This is great work!

closed 2021-11-02 by ooxi

eyal0, 2021-11-02:
Thanks!  This was kind of a tricky one!  I had to dig into gdk some.  I hope that they don't deprecate the functions that I used in the end.

ooxi, 2021-11-02:
> Thanks! This was kind of a tricky one! I had to dig into gdk some. I hope that they don't deprecate the functions that I used in the end.

That would be really unfortunate :D but it will at least buy us a couple of years, I hope

PR #47 (2021-11-02): Replace gdk_pixbuf_new_from_inline with gdk_pixbuf_new_from_data
This removes 4 compiler warnings.  The icons themselves are unchanged.

This fixes #46.