gerbv/gerbv #56: Gerbv RS-274X format aperture macro variables out-of-bounds write vulnerability (TALOS-2021-1404)

CiscoTalos, 2021-11-03:
TALOS-2021-1404   
CVE-2021-40393   

Gerbv RS-274X format aperture macro variables out-of-bounds write vulnerability

### Summary

An out-of-bounds write vulnerability exists in the RS-274X aperture macro variables handling functionality of Gerbv 2.7.0 and dev (commit b5f1eacd). A specially-crafted gerber file can lead to code execution. An attacker can provide a malicious file to trigger this vulnerability.

### Tested Versions

Gerbv 2.7.0   
Gerbv dev  (commit b5f1eacd)   
Gerbv forked dev (commit 71493260)   

### Product URLs

[https://sourceforge.net/projects/gerbv/](https://sourceforge.net/projects/gerbv/)

### CVSSv3 Score

10.0 - CVSS:3.0/AV:N/AC:L/PR:N/UI:N/S:C/C:H/I:H/A:H

### CWE

CWE-119 - Improper Restriction of Operations within the Bounds of a Memory Buffer

### Details

Gerbv is an open-source software that allows to view RS-274X Gerber files, Excellon drill files and pick-n-place files. These file formats are used in industry to describe the layers of a printed circuit board and are a core part of the manufacturing process.

Some PCB (printed circuit board) manufacturers use software like Gerbv in their web interfaces as a tool to convert Gerber (or other supported) files into images. Users can upload gerber files to the manufacturer website, which are converted to an image to be displayed in the browser, so that users can verify that what has been uploaded matches their expectations. Gerbv can do such conversions using the `-x` switch (export). For this reason, we consider this software as reachable via network without user interaction or privilege requirements.

Gerbv uses the function `gerbv_open_image` to open files. In this advisory we're interested in the RS-274X file-type.

    int
    gerbv_open_image(gerbv_project_t *gerbvProject, char *filename, int idx, int reload,
                    gerbv_HID_Attribute *fattr, int n_fattr, gboolean forceLoadFile)
    {
        ...        
        dprintf("In open_image, about to try opening filename = %s\n", filename);
        
        fd = gerb_fopen(filename);
        if (fd == NULL) {
            GERB_COMPILE_ERROR(_("Trying to open \"%s\": %s"),
                            filename, strerror(errno));
            return -1;
        }
        ...
        if (gerber_is_rs274x_p(fd, &foundBinary)) {                                 // [1]
            dprintf("Found RS-274X file\n");
            if (!foundBinary || forceLoadFile) {
                    /* figure out the directory path in case parse_gerb needs to
                     * load any include files */
                    gchar *currentLoadDirectory = g_path_get_dirname (filename);
                    parsed_image = parse_gerb(fd, currentLoadDirectory);            // [2]
                    g_free (currentLoadDirectory);
            }
        }
        ...

A file is considered of type "RS-274X" if the function `gerber_is_rs274x_p` [1] returns true. When true, the `parse_gerb` is called [2] to parse the input file. Let's first look at the requirements that we need to satisfy to have an input file be recognized as an RS-274X file:

    gboolean
    gerber_is_rs274x_p(gerb_file_t *fd, gboolean *returnFoundBinary) 
    {
        ...
        while (fgets(buf, MAXL, fd->fd) != NULL) {
            dprintf ("buf = \"%s\"\n", buf);
            len = strlen(buf);
[... 376 lines cut ...]
        #2 0xf795e5b8 in parse_rs274x src/gerber.c:1637
        #3 0xf7951ad8 in gerber_parse_file_segment src/gerber.c:243
        #4 0xf7957660 in parse_gerb src/gerber.c:768
        #5 0xf796d690 in gerbv_open_image src/gerbv.c:526
        #6 0xf796b03d in gerbv_open_layer_from_filename_with_color src/gerbv.c:249
        #7 0x565bdcab in main src/main.c:929
        #8 0xf6b6cf20 in __libc_start_main (/lib/i386-linux-gnu/libc.so.6+0x18f20)
        #9 0x5657bc10  (gerbv+0x12c10)

    AddressSanitizer can not provide additional info.
    SUMMARY: AddressSanitizer: SEGV src/gerber.c:1944 in simplify_aperture_macro
    ==5293==ABORTING

### Credit

Discovered by Claudio Bozzato of Cisco Talos.
     
https://talosintelligence.com/vulnerability_reports/

### Timeline

2021-11-03 - Vendor Disclosure   
None - Public Release

[TALOS-2021-1404 - Gerbv_RS-274X_format_aperture_macro_variables_out-of-bounds_write_vulnerability.txt](https://github.com/gerbv/gerbv/files/7468965/TALOS-2021-1404.-.Gerbv_RS-274X_format_aperture_macro_variables_out-of-bounds_write_vulnerability.txt)

CiscoTalos, 2021-11-03:
We can email the poc files once you acknowledge since they are unable to be attached in this issue

eyal0, 2021-11-03:
@CiscoTalos Looks like you are able to run gerbv with ASAN.  Can you send what changes you made to the the build so that it would run with ASAN?  I would like to do that, too.

CiscoTalos, 2021-11-03:
No changes to sources are needed, we compiled using `./configure CFLAGS="-fsanitize=address" --disable-shared --disable-update-desktop-database` on Ubuntu 18.04 32 bit.

CiscoTalos, 2021-11-04:
[TALOS-2021-1404.poc.zip](https://github.com/gerbv/gerbv/files/7477457/TALOS-2021-1404.poc.zip)

ooxi, 2021-12-06:
Fix published in release candidate [2.8.2-rc.1](https://github.com/gerbv/gerbv/releases/tag/v2.8.2-rc.1), backported to [2.7.3-rc.1](https://github.com/gerbv/gerbv/releases/tag/v2.7.3-rc.1)

ooxi, 2021-12-19:
Closing this issue since a fix has been published in release [2.8.2](https://github.com/gerbv/gerbv/releases/tag/v2.8.2), backported to [2.7.3](https://github.com/gerbv/gerbv/releases/tag/v2.7.3); feel free to reopen if necessary.

Thank you very much @CiscoTalos for reporting this issue! Please ping us as soon as you publicly disclose the report so we can update the references in our documentation.

closed 2021-12-19 by ooxi