gerbv/gerbv #57: Gerbv RS-274X aperture macro outline primitive integer overflow vulnerability (TALOS-2021-1405)

CiscoTalos, 2021-11-03:
TALOS-2021-1405   
CVE-2021-40394   

Gerbv RS-274X aperture macro outline primitive integer overflow vulnerability

### Summary

An integer overflow vulnerability exists in the RS-274X aperture macro outline primitive functionality of Gerbv 2.7.0 and dev (commit b5f1eacd). A specially-crafted gerber file can lead to code execution. An attacker can provide a malicious file to trigger this vulnerability.

### Tested Versions

Gerbv 2.7.0   
Gerbv dev  (commit b5f1eacd)   
Gerbv forked dev (commit 71493260)   

### Product URLs

[https://sourceforge.net/projects/gerbv/](https://sourceforge.net/projects/gerbv/)

### CVSSv3 Score

10.0 - CVSS:3.0/AV:N/AC:L/PR:N/UI:N/S:C/C:H/I:H/A:H

### CWE

CWE-190 - Integer Overflow or Wraparound

### Details

Gerbv is an open-source software that allows to view RS-274X Gerber files, Excellon drill files and pick-n-place files. These file formats are used in industry to describe the layers of a printed circuit board and are a core part of the manufacturing process.

Some PCB (printed circuit board) manufacturers use software like Gerbv in their web interfaces as a tool to convert Gerber (or other supported) files into images. Users can upload gerber files to the manufacturer website, which are converted to an image to be displayed in the browser, so that users can verify that what has been uploaded matches their expectations. Gerbv can do such conversions using the `-x` switch (export). For this reason, we consider this software as reachable via network without user interaction or privilege requirements.

Gerbv uses the function `gerbv_open_image` to open files. In this advisory we're interested in the RS-274X file-type.

    int
    gerbv_open_image(gerbv_project_t *gerbvProject, char *filename, int idx, int reload,
                    gerbv_HID_Attribute *fattr, int n_fattr, gboolean forceLoadFile)
    {
        ...        
        dprintf("In open_image, about to try opening filename = %s\n", filename);
        
        fd = gerb_fopen(filename);
        if (fd == NULL) {
            GERB_COMPILE_ERROR(_("Trying to open \"%s\": %s"),
                            filename, strerror(errno));
            return -1;
        }
        ...
        if (gerber_is_rs274x_p(fd, &foundBinary)) {                                 // [1]
            dprintf("Found RS-274X file\n");
            if (!foundBinary || forceLoadFile) {
                    /* figure out the directory path in case parse_gerb needs to
                     * load any include files */
                    gchar *currentLoadDirectory = g_path_get_dirname (filename);
                    parsed_image = parse_gerb(fd, currentLoadDirectory);            // [2]
                    g_free (currentLoadDirectory);
            }
        }
        ...

A file is considered of type "RS-274X" if the function `gerber_is_rs274x_p` [1] returns true. When true, the `parse_gerb` is called [2] to parse the input file. Let's first look at the requirements that we need to satisfy to have an input file be recognized as an RS-274X file:

    gboolean
    gerber_is_rs274x_p(gerb_file_t *fd, gboolean *returnFoundBinary) 
    {
        ...
        while (fgets(buf, MAXL, fd->fd) != NULL) {
            dprintf ("buf = \"%s\"\n", buf);
            len = strlen(buf);
[... 429 lines cut ...]
      Stack after return:      f5
      Stack use after scope:   f8
      Global redzone:          f9
      Global init order:       f6
      Poisoned by user:        f7
      Container overflow:      fc
      Array cookie:            ac
      Intra object redzone:    bb
      ASan internal:           fe
      Left alloca redzone:     ca
      Right alloca redzone:    cb
    ==9184==ABORTING

### Credit

Discovered by Claudio Bozzato of Cisco Talos.
     
https://talosintelligence.com/vulnerability_reports/

### Timeline

2021-11-03 - Vendor Disclosure   
None - Public Release

[TALOS-2021-1405 - Gerbv_RS-274X_aperture_macro_outline_primitive_integer_overflow_vulnerability.txt](https://github.com/gerbv/gerbv/files/7469006/TALOS-2021-1405.-.Gerbv_RS-274X_aperture_macro_outline_primitive_integer_overflow_vulnerability.txt)

ooxi, 2021-11-04:
Again, thank you very much for this detailed report! We will take some time to analyze the issue and propose a fix.

ooxi, 2021-11-04:
Signed integer operations is a mine field. While this instance can be fixed quite easily, we should invest in a global solution like utilizing a library which either traps or clamps overflows.

ooxi, 2021-11-04:
I proposed a possible hotfix for this issue in pull request #61

CiscoTalos, 2021-11-04:
Here is the poc. Sorry for the delay, we'll attach pocs to the other bugs as well.
[TALOS-2021-1405.poc.zip](https://github.com/gerbv/gerbv/files/7477448/TALOS-2021-1405.poc.zip)

ooxi, 2021-11-24:
Fixed by pull request #61, released as [v2.8.1](https://github.com/gerbv/gerbv/releases/tag/v2.8.1) and backported to [v2.7.2](https://github.com/gerbv/gerbv/releases/tag/v2.7.2).

@CiscoTalos thank you very much for your report! Feel free to reopen this issue if you have any additional input. Please also provide an information when you have publicly disclosed this issue so we can update our links.

closed 2021-11-24 by ooxi