gerbv/gerbv #60: Remove all clang compile-time warnings

eyal0, 2021-11-04:
Currently, CI only builds using gcc.  If we build using clang, -Werror reports other warnings.  We should fix those, too.

We can either have a single CI that will build, make clean, and then build again with clang.  Alternatively, we can have the CI launch two different jobs and do it separately.  This is a little more complex to write but it would run faster, because it would be in parallel, and maybe it's a cleaner solution because we wouldn't be relying on `make clean`.

If we do split the builds, we would want to only do the deploy stuff if *both* CI jobs succeed.  So probably we'd have two CI jobs, one doing gcc, one clang.  And when both succeed, then launch the job that does all the releases and deploy github pages and stuff.

eyal0, 2021-11-04:
#45 should be completed first.

ooxi, 2021-11-04:
If we find a mechanism to split the build into multiple jobs but do deployment only if all steps succeed, then I would prefer that solution. We could then also push the different distribution builds into separate jobs to speed up the build even further.

eyal0, 2021-11-05:
So let's see how we'd like it to work.

First, we want to build and test.  Each build+test combo would be in a unique job and the different jobs might be, like, unix_gcc, unix_clang, macos_gcc, windows_gcc, etc.  For now we can just have unix gcc and add more as we go.

Then, only if all the previous ones pass, then what?  Do we do all the npx stuff in a single job?  Or does each job do its own npx?  npx is for making npm modules or something?  I dunno much about it.

And there is also the step to update the github.io page.  Where does that go?

If we can figure out which jobs we want to do and which parts are dependent on which other parts, I can string it all together.  I've done it before for a different project:

[image]

A dozen builds in parallel and then only when they all complete successfully, the last two steps run.  There's a little overhead for launching a job so it doesn't make sense to chain together jobs that are less than like, 2 minutes or so.

If one job is creating outputs that another job needs, it is possible to communicate between jobs.  Basically you save data to a file and then read it back in a subsequent job.

So it's pretty flexible once we figure out what we want to do!