gerbv/gerbv #63: Please a more common way of tagging versions

tijuca, 2021-11-07:
Thanks for working further on gerbv!

I'd like to request a wish for further releases.
Please use a common kind of tagging and versioning of release tarballs.

Currently (in contrary to the SF releases) the tagged versions are named `gerbv-2-8-0-RC-1` or `gerbv-2-7-1-RELEASE`. That makes it unnecessary hard to adjust the watch logic from an effort perspective within Debian which end in manual version tracking. Doing a tagging this way is quite uncommon.

So I hereby like to request to use the one of the common ways to tag versions line `2.8.0-rc1` or also prefixing a letter 'v' so `v2.7.1` could be used. That would help a lot to let me use the watch daemon on Debian which informs me about new upstream versions.

ooxi, 2021-11-08:
That is a great suggestion, thank you! To make sure I understand you correctly, you would prefer

* `v2.8.0-rc.1`
* `v2.7.1`
* `v2.7.0`

ooxi, 2021-11-08:
If nobody objects, I will go ahead and create the following tags:

* [`v2.8.0-rc.1`](https://github.com/gerbv/gerbv/commit/41574dc1e8b0b36675b0192ee412d632ab78d0b7)
* [`v2.7.1`](https://github.com/gerbv/gerbv/commit/96e82fe9c1f235e2b8eb9422ab4659c188d96b75)
* [`v2.7.0`](https://github.com/gerbv/gerbv/commit/3c70dc88260086e303ff8722dc9e6aec73e3b7e2)
* [`v2.6.2`](https://github.com/gerbv/gerbv/commit/85ca39ad30884d4d4805c7a230da4d749428ddb5)
* [`v2.6.0`](https://github.com/gerbv/gerbv/commit/c8b252486351e32d7644acbfa73f107a2b26b2ea)
* [`v2.5.0`](https://github.com/gerbv/gerbv/commit/909d57ceb0380c5d4f6c0b2a5aad1acbaeeee5dd)
* [`v2.4.0`](https://github.com/gerbv/gerbv/commit/67e1c901d52bba817bf45adab9a69e507094dd10)

@tijuca please confirm that those will be sufficient for your use case. We did not come up with the old naming scheme, that was carried over from gEDA. Thus I want to make sure if I understand you correctly.

tijuca, 2021-11-08:
@ooxi 

I'm fine with tagging strings, except for the RC, there is no need to include a point between the characters `rc` and the number of the release candidate, so I'd do simply `v2.7.8-rc1` and so on. Even `v2.7.8rc1` would be fine, the watch integration is detecting both as release candidate and we can adjust the version string for packaging.

Thank you for rethinking the tagging! Makes my life as package maintainer easier.

ooxi, 2021-11-08:
Thank you for the feedback @tijuca! We strive to make maintainer's life easier since they remove a lot of burden from us, too :-)

With regards to RC, I don't think we will change this. We discussed in [PR#58](https://github.com/gerbv/gerbv/pull/58#issuecomment-960596073) to use the syntax popularized by [semver](https://semver.org/spec/v2.0.0-rc.2.html).

Nevertheless, I hope this does not impact you too much, since you will most likely be more interested in releases, not release candidates.

tijuca, 2021-11-08:
It's up to the project maintainers to make decisions for their project, so you are free to do what ever you like.

And yes, we are mostly interested in the main and final releases. That not mean we will not package RC versions. This depends on how intrusive a new release might be, so sometimes it's helpful to package also a RC.

ooxi, 2021-11-10:
@tijuca in addition to the commits lined out above, I have also found [2.6.2](https://github.com/gerbv/gerbv/tree/v2.6.2).

Hope this helps. Feel free to reopen this issue if you have additional input!

closed 2021-11-10 by ooxi