gerbv/gerbv #64: New command line arguments

ferdymercury, 2021-11-08:
It would be really nice to have the ability to control via CLI the following items when starting the program:
- Orientation of a given layer (Modify Orientation)
- Drawing mode (Fast, XOR, ...)

This is needed for: https://github.com/dennevi/GrbDiff/issues/1

ooxi, 2021-11-08:
Thank you for your suggestion! Modifying the default drawing mode should be quite easy using command line options. With regards to the orientation of given layers, wouldn't it be better to do this via gerbv project files?

neinseg, 2022-01-22:
I recently had a similar issue with the excellon file number format options. I was able to override these options though a project file, but a command line option would have been easier. My use case was exporting a single excellon file to svg.