gerbv/gerbv #65: save project does not keep orientation

ferdymercury, 2021-11-08:
When saving a project, closing and reopening, the orientation set on a layer via Right click, Modify Orientation, is lost.

I am using gerbv 2.6.1 (official Ubuntu 18 package).

ooxi, 2021-11-08:
Thank you for reporting this! While 2.6.1 is quite old and no longer supported, I'm fairly certain this behaviour is still present. If I remember correctly, I was bit by it, too.

ferdymercury, 2021-11-08:
Thanks to you for reviving gerbv ;)

Yes, I am thinking of compiling by source to get the latest.

tijuca, 2021-11-15:
Hi, within Debian we have also a bug report about not saved project settings, might be the same as already reported from @ferdymercury 

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=727692

Rix, 2021-11-26:
> Yes, I am thinking of compiling by source to get the latest.

I've tried the latest source code, but it seems that bug does not exists in the latest version. I was able to reload modified layer 'as modified' from the saved project file. Were you able to try on v.2.8.1 release ?

ferdymercury, 2021-11-26:
Thanks for the hint. I compiled from source today and indeed it's working well !

closed 2021-11-26 by ferdymercury