gerbv/gerbv #79: Security advisory TALOS-2021-1413

CiscoTalos, 2021-11-22:
TALOS-2021-1413   
CVE-2021-40400   

Gerbv RS-274X aperture macro outline primitive out-of-bounds read vulnerability

### Summary

An out-of-bounds read vulnerability exists in the RS-274X aperture macro outline primitive functionality of Gerbv 2.7.0 and dev (commit b5f1eacd) and the forked version of Gerbv (commit d7f42a9a). A specially-crafted gerber file can lead to information disclosure. An attacker can provide a malicious file to trigger this vulnerability.

### Tested Versions

Gerbv 2.7.0   
Gerbv dev  (commit b5f1eacd)   
Gerbv forked dev (commit d7f42a9a)   

### Product URLs

[https://sourceforge.net/projects/gerbv/](https://sourceforge.net/projects/gerbv/)

### CVSSv3 Score

9.3 - CVSS:3.0/AV:N/AC:L/PR:N/UI:N/S:C/C:L/I:N/A:H

### CWE

CWE-119 - Improper Restriction of Operations within the Bounds of a Memory Buffer

### Details

Gerbv is an open-source software that allows to view RS-274X Gerber files, Excellon drill files and pick-n-place files. These file formats are used in industry to describe the layers of a printed circuit board and are a core part of the manufacturing process.

Some PCB (printed circuit board) manufacturers use software like Gerbv in their web interfaces as a tool to convert Gerber (or other supported) files into images. Users can upload gerber files to the manufacturer website, which are converted to an image to be displayed in the browser, so that users can verify that what has been uploaded matches their expectations. Gerbv can do such conversions using the `-x` switch (export). Moreover, gerbv can be compiled and used as a shared library. For these reasons, we consider this software as reachable via network without user interaction or privilege requirements.

Gerbv uses the function `gerbv_open_image` to open files. In this advisory we're interested in the RS-274X file-type.

    int
    gerbv_open_image(gerbv_project_t *gerbvProject, char *filename, int idx, int reload,
                    gerbv_HID_Attribute *fattr, int n_fattr, gboolean forceLoadFile)
    {
        ...        
        dprintf("In open_image, about to try opening filename = %s\n", filename);
        
        fd = gerb_fopen(filename);
        if (fd == NULL) {
            GERB_COMPILE_ERROR(_("Trying to open \"%s\": %s"),
                            filename, strerror(errno));
            return -1;
        }
        ...
        if (gerber_is_rs274x_p(fd, &foundBinary)) {                                 // [1]
            dprintf("Found RS-274X file\n");
            if (!foundBinary || forceLoadFile) {
                    /* figure out the directory path in case parse_gerb needs to
                     * load any include files */
                    gchar *currentLoadDirectory = g_path_get_dirname (filename);
                    parsed_image = parse_gerb(fd, currentLoadDirectory);            // [2]
                    g_free (currentLoadDirectory);
            }
        }
        ...

A file is considered of type "RS-274X" if the function `gerber_is_rs274x_p` [1] returns true. When true, the `parse_gerb` is called [2] to parse the input file. Let's first look at the requirements that we need to satisfy to have an input file be recognized as an RS-274X file:

    gboolean
    gerber_is_rs274x_p(gerb_file_t *fd, gboolean *returnFoundBinary) 
    {
        ...
        while (fgets(buf, MAXL, fd->fd) != NULL) {
            dprintf ("buf = \"%s\"\n", buf);
            len = strlen(buf);
[... 419 lines cut ...]
      Stack mid redzone:       f2
      Stack right redzone:     f3
      Stack after return:      f5
      Stack use after scope:   f8
      Global redzone:          f9
      Global init order:       f6
      Poisoned by user:        f7
      Container overflow:      fc
      Array cookie:            ac
      Intra object redzone:    bb
      ASan internal:           fe
      Left alloca redzone:     ca
      Right alloca redzone:    cb
    ==267==ABORTING

### Credit

Discovered by Claudio Bozzato of Cisco Talos.
     
https://talosintelligence.com/vulnerability_reports/

### Timeline

None - Vendor Disclosure   
None - Public Release

CiscoTalos, 2021-11-22:
poc: [aperture_macro_parameters_oobr.zip](https://github.com/gerbv/gerbv/files/7584971/aperture_macro_parameters_oobr.zip)

CiscoTalos, 2021-12-20:
Any updates to this issue? Out of bounds issues (56 and 82) have proposed fixes

CiscoTalos, 2022-01-28:
Does this issue have any updates regarding fix?

CiscoTalos, 2022-02-17:
Please advise if there is a fix for this issue. The issue is beyond 90 days and subject to public disclosure on our end.

CiscoTalos, 2022-02-22:
Should this be disclosed along with #80 without a fix?

CiscoTalos, 2022-02-24:
To date, this issue is beyond 90 days and subject to public disclosure. If there are no updates to this issue, we will publish along with #80 in the first week of March

ooxi, 2022-02-27:
@CiscoTalos please publish

ooxi, 2022-05-29:
Proposed fix in pull request #124

ooxi, 2022-05-31:
Pull request #124 merged as caa6560d5d683f827c672fd5e380f89a8ef632b6 and will be included in the upcoming release

ooxi, 2022-06-04:
@CiscoTalos Fix included in release [2.9.2](https://github.com/gerbv/gerbv/releases/tag/v2.9.2). Thanks for your support!

closed 2022-06-04 by ooxi