gerbv/gerbv #81: Secrurity Advisory (TALOS-2021-1415)

CiscoTalos, 2021-11-24:
Gerbv RS-274X aperture definition tokenization use-after-free vulnerability

### Summary

A use-after-free vulnerability exists in the RS-274X aperture definition tokenization functionality of Gerbv 2.7.0 and dev (commit b5f1eacd) and Gerbv forked 2.7.1. A specially-crafted gerber file can lead to code execution. An attacker can provide a malicious file to trigger this vulnerability.
### Tested Versions

Gerbv 2.7.0   
Gerbv forked 2.7.1   
Gerbv dev  (commit b5f1eacd)   

### Product URLs

[https://sourceforge.net/projects/gerbv/](https://sourceforge.net/projects/gerbv/)

### CVSSv3 Score

10.0 - CVSS:3.0/AV:N/AC:L/PR:N/UI:N/S:C/C:H/I:L/A:H

### CWE

CWE-252 - Unchecked Return Value

### Details

Gerbv is an open-source software that allows users to view RS-274X Gerber files, Excellon drill files and pick-n-place files. These file formats are used in industry to describe the layers of a printed circuit board and are a core part of the manufacturing process.

Some PCB (printed circuit board) manufacturers use software like Gerbv in their web interfaces as a tool to convert Gerber (or other supported) files into images. Users can upload gerber files to the manufacturer website, which are converted to an image to be displayed in the browser, so that users can verify that what has been uploaded matches their expectations. Gerbv can do such conversions using the `-x` switch (export). For this reason, we consider this software as reachable via network without user interaction or privilege requirements.

Gerbv uses the function `gerbv_open_image` to open files. In this advisory we're interested in the RS-274X file-type.

    int
    gerbv_open_image(gerbv_project_t *gerbvProject, char *filename, int idx, int reload,
                    gerbv_HID_Attribute *fattr, int n_fattr, gboolean forceLoadFile)
    {
        ...        
        dprintf("In open_image, about to try opening filename = %s\n", filename);
        
        fd = gerb_fopen(filename);
        if (fd == NULL) {
            GERB_COMPILE_ERROR(_("Trying to open \"%s\": %s"),
                            filename, strerror(errno));
            return -1;
        }
        ...
        if (gerber_is_rs274x_p(fd, &foundBinary)) {                                 // [1]
            dprintf("Found RS-274X file\n");
            if (!foundBinary || forceLoadFile) {
                    /* figure out the directory path in case parse_gerb needs to
                     * load any include files */
                    gchar *currentLoadDirectory = g_path_get_dirname (filename);
                    parsed_image = parse_gerb(fd, currentLoadDirectory);            // [2]
                    g_free (currentLoadDirectory);
            }
        }
        ...

A file is considered of type "RS-274X" if the function `gerber_is_rs274x_p` [1] returns true. When true, the `parse_gerb` is called [2] to parse the input file. Let's first look at the requirements that we need to satisfy to have an input file be recognized as an RS-274X file:

    gboolean
    gerber_is_rs274x_p(gerb_file_t *fd, gboolean *returnFoundBinary) 
    {
        ...
        while (fgets(buf, MAXL, fd->fd) != NULL) {
            dprintf ("buf = \"%s\"\n", buf);
            len = strlen(buf);
        
            /* First look through the file for indications of its type by
             * checking that file is not binary (non-printing chars and white 
             * spaces)
[... 458 lines cut ...]

Include an external file containing

    %ADD21

This triggers the `strtok` issue, since there's no `*` till the end of this file. Note that there might be a way to do the same without the `%IF` directive, however we couldn't find a trivial way around using `M02*` below and force the rending of the image at the same time.

    D21*
    X400000Y400000D03*
    M02*

Finally, use the tool 21, which will use whatever macro has been matched depending on heap contents (it might match one of "-V", "\_V", "=V", however in the provided PoC will likely match nothing unless run in gdb). If any match happen one of the triangles defined by the macro will be rendered in the image. By looking at which triangle has been printed, we will know which byte was in memory, hence the information leak.

### Credit

Discovered by Claudio Bozzato of Cisco Talos.
     
https://talosintelligence.com/vulnerability_reports/

### Timeline

2021-11-24 - Vendor Disclosure   
None - Public Release

[TALOS-2021-1415 - Gerbv_RS-274X_aperture_definition_tokenization_use-after-free_vulnerability.txt](https://github.com/gerbv/gerbv/files/7596124/TALOS-2021-1415.-.Gerbv_RS-274X_aperture_definition_tokenization_use-after-free_vulnerability.txt)

CiscoTalos, 2021-11-24:
[TALOS-2021-1415.zip](https://github.com/gerbv/gerbv/files/7596368/TALOS-2021-1415.zip)

ooxi, 2021-12-19:
Proposed fix 3bf22983dd36d50f3e769d9bca447df2b2f42eb0 in pull request #88.

Does this issue already have a CVE identifier assigned?

CiscoTalos, 2021-12-20:
CVE-2021-40401 assigned

CiscoTalos, 2022-01-06:
The proposed fix is good. Do you have a release date planned?

ooxi, 2022-01-27:
@CiscoTalos The proposed fix has been merged into `main`. Please continue with public disclosure

ooxi, 2022-05-18:
@CiscoTalos Fix included in release [2.9.0](https://github.com/gerbv/gerbv/releases/tag/v2.9.0)

closed 2022-05-18 by ooxi