gerbv/gerbv #82: Security Advisory (TALOS-2021-1417)

CiscoTalos, 2021-11-24:
TALOS-2021-1417   
CVE-2021-40403   

Gerbv pick-and-place rotation parsing use of uninitialized variable vulnerability

### Summary

An information disclosure vulnerability exists in the pick-and-place rotation parsing functionality of Gerbv 2.7.0 and dev (commit b5f1eacd), and Gerbv forked 2.8.0. A specially-crafted pick-and-place file can exploit the missing initialization of a structure to leak memory contents. An attacker can provide a malicious file to trigger this vulnerability.

### Tested Versions

Gerbv 2.7.0   
Gerbv forked 2.8.0   
Gerbv dev  (commit b5f1eacd)   

### Product URLs

[https://sourceforge.net/projects/gerbv/](https://sourceforge.net/projects/gerbv/)

### CVSSv3 Score

5.8 - CVSS:3.0/AV:N/AC:L/PR:N/UI:N/S:C/C:L/I:N/A:N

### CWE

CWE-456 - Missing Initialization of a Variable

### Details

Gerbv is an open-source software that allows users to view RS-274X Gerber files, Excellon drill files and pick-n-place files. These file formats are used in industry to describe the layers of a printed circuit board and are a core part of the manufacturing process.

Some PCB (printed circuit board) manufacturers use software like Gerbv in their web interfaces as a tool to convert Gerber (or other supported) files into images. Users can upload gerber files to the manufacturer website, which are converted to an image to be displayed in the browser, so that users can verify that what has been uploaded matches their expectations. Gerbv can do such conversions using the `-x` switch (export). For this reason, we consider this software as reachable via network without user interaction or privilege requirements.

Gerbv uses the function `gerbv_open_image` to open files. In this advisory we're interested in the pick-and-place file-type.

    int
    gerbv_open_image(gerbv_project_t *gerbvProject, char *filename, int idx, int reload,
                    gerbv_HID_Attribute *fattr, int n_fattr, gboolean forceLoadFile)
    {
        ...
        dprintf("In open_image, about to try opening filename = %s\n", filename);

        fd = gerb_fopen(filename);
        if (fd == NULL) {
            GERB_COMPILE_ERROR(_("Trying to open \"%s\": %s"),
                            filename, strerror(errno));
            return -1;
        }
        ...
        } else if (pick_and_place_check_file_type(fd, &foundBinary)) {                              // [1]
            dprintf("Found pick-n-place file\n");
            if (!foundBinary || forceLoadFile) {
                    if (!reload) {
                            pick_and_place_parse_file_to_images(fd, &parsed_image, &parsed_image2); // [2]
        ...

A file is considered of type "pick-and-place" if the function `pick_and_place_check_file_type` [1] returns true. When true, `pick_and_place_parse_file_to_images` is called [2] to parse the input file. Let's first look at the requirements that we need to satisfy to have an input file be recognized as an pick-and-place file:

    gboolean
    pick_and_place_check_file_type(gerb_file_t *fd, gboolean *returnFoundBinary)
    {
        ...
        while (fgets(buf, MAXL, fd->fd) != NULL) {
            len = strlen(buf);
         
            /* First look through the file for indications of its type */
            
            /* check for non-binary file */
            for (i = 0; i < len; i++) {                                             // [3]
                if (!isprint((int) buf[i]) && (buf[i] != '\r') && 
[... 231 lines cut ...]
    ==863== Conditional jump or move depends on uninitialised value(s)
    ==863==    at 0x159587: pick_and_place_parse_file (pick-and-place.c:373)
    ==863==    by 0x15AB5B: pick_and_place_parse_file_to_images (pick-and-place.c:790)
    ==863==    by 0x156C92: gerbv_open_image (gerbv.c:538)
    ==863==    by 0x1561B0: gerbv_open_layer_from_filename_with_color (gerbv.c:249)
    ==863==    by 0x12D6C9: main (main.c:932)
    ==863==
    ==863== Conditional jump or move depends on uninitialised value(s)
    ==863==    at 0x1595A7: pick_and_place_parse_file (pick-and-place.c:373)
    ==863==    by 0x15AB5B: pick_and_place_parse_file_to_images (pick-and-place.c:790)
    ==863==    by 0x156C92: gerbv_open_image (gerbv.c:538)
    ==863==    by 0x1561B0: gerbv_open_layer_from_filename_with_color (gerbv.c:249)
    ==863==    by 0x12D6C9: main (main.c:932)
    ==863==

### Credit

Discovered by Claudio Bozzato of Cisco Talos.
     
https://talosintelligence.com/vulnerability_reports/

### Timeline

2021-11-24 - Vendor Disclosure   
None - Public Release

CiscoTalos, 2021-11-24:
[TALOS-2021-1417.zip](https://github.com/gerbv/gerbv/files/7596403/TALOS-2021-1417.zip)

ooxi, 2021-12-16:
Thank you very much for your patience. We will try to provide a patch with one of the upcoming releases.

ooxi, 2021-12-19:
Proposed fix 387f07b163cc30cd95e9bedf53bc07e7b38cc318 in pull request #85

CiscoTalos, 2022-01-06:
The proposed fix is good. Please confirm release date and we will coordinate public disclosure on our end.

CiscoTalos, 2022-01-18:
Is there a planned disclosure release date for this issue?

ooxi, 2022-01-27:
> Is there a planned disclosure release date for this issue?

Please disclose the bug as is. I am hesitant to merge this fix since it breaks a test case. But that's no reason to postpone disclosure.

ooxi, 2022-05-21:
Proposed fix merged as c32c6f9c0b5d3b0ecc33de21d8532de6c2df5878 in pull request #85 and will be included in next release.

ooxi, 2022-05-29:
@CiscoTalos Fix included in release [2.9.1](https://github.com/gerbv/gerbv/releases/tag/v2.9.1)

closed 2022-05-29 by ooxi