gerbv/gerbv #83: Unsupported codes in Route files

htot, 2021-12-02:
When importing I get:
NAME-SlotHoles-NonPlated.TXT"
Unsupported G01 (linear mode) code at line 25 in file "NAME-SlotHoles-NonPlated.TXT"
Unsupported M16 (retract with clamping) code found at line 26 in file "NAME-SlotHoles-NonPlated.TXT"
Unsupported G00 (rout mode) code at line 27 in file "NAME-SlotHoles-NonPlated.TXT"
Unsupported M15 (Z-axis rout position) code found at line 28 in file "NAME-SlotHoles-NonPlated.TXT"
Unsupported G01 (linear mode) code at line 29 in file "NAME-SlotHoles-NonPlated.TXT"
Unsupported M16 (retract with clamping) code found at line 30 in file "NAME-SlotHoles-NonPlated.TXT"
Unsupported M17 (retract without clamping) code found at line 31 in file "NAME-SlotHoles-NonPlated.TXT"

ooxi, 2021-12-06:
Could you please paste the source file?

htot, 2021-12-07:
[NAME-SlotHoles-NonPlated.TXT](https://github.com/gerbv/gerbv/files/7668823/NAME-SlotHoles-NonPlated.TXT)

ooxi, 2021-12-07:
Thank you for bringing this to our attention. Currently only a [limited number for G-codes](https://github.com/gerbv/gerbv/blob/main/src/drill.c#L454) are implemented for parsing Excellon files:

* `DRILL_G_DRILL` (`G05`): drill mode
* `DRILL_G_SLOT` (`G85`): canned slot
* `DRILL_G_ABSOLUTE` (`G90`): absolute input mode
* `DRILL_G_INCREMENTAL` (`G91`): incremental input mode
* `DRILL_G_ZEROSET` (`G93`): zero set

We are very welcome to patches however!

closed 2021-12-07 by ooxi