gerbv/gerbv #91: gerbv should probably complain about unterminated G04 comments.

neinseg, 2022-01-23:
Sample file attached. This file contains a mid-file G04 comment with the trailing '*' missing. gerbv silently parses this as a two-line G04 comment, leading to incorrect rendering. It should probably complain about the unterminated comment instead.

```
G54D17*
X00953Y00818D03*
X00953Y00618D03*
X00953Y00418D03*
G04 next file*
G04 MADE WITH FRITZING*
G04 WWW.FRITZING.ORG*
G04 DOUBLE SIDED*
G04 HOLES PLATED*
G04 CONTOUR ON CENTER OF CONTOUR VECTOR*
G90*
G04 skipping 70
D28*
X01338Y00039D02*
X01751Y01768D03*
X01851Y01768D03*
X01951Y01768D03*
X02051Y01768D03*
D29*
X01551Y01418D03*
X01551Y01118D03*
```

The four holes at Y=1.768 get rendered using the rectangular D17 aperture instead of the circular D28 aperture.

[sample_gerber.txt](https://github.com/gerbv/gerbv/files/7921155/sample_gerber.txt)

eyal0, 2022-01-24:
You could perhaps issue a warning but you should not issue an error.  The only characters that cannot be in a comment are the percent character `%` and the asterisk `*`.

So gerbv is doing the right thing here.  A warning would just be a sort of courtesy to the user.

eyal0, 2022-01-24:
https://www.ucamco.com/files/downloads/file_en/416/the-gerber-layer-format-specification-revision-2021-02_en.pdf?1109f5f3efc7ef822119bd893d685572 section 4.1 describes comments as containing a "string".  3.4.3 describes strings.

neinseg, 2022-01-29:
I totally agree that this is invalid gerber. But since that particular generator implementation was faulty, this problem exists in many gerber files floating around on github. So a clear warning would be useful, since the alternative is a silent corruption.

eyal0, 2022-02-01:
Looks like you're not the first one to notice this problem!  https://github.com/gerbv/gerbv/blob/main/src/gerber.c#L1008

eyal0, 2022-02-01:
Tell me, how does this look to you:

[image]

[image]

When you open a file with messages, the messages tab opens by default.

jaseg, 2022-02-03:
Looks good to me.

PR #92 (2022-02-04): Emit warning when G04 includes a CR or LF.
This fixes #91
closed 2022-02-04 by eyal0