gerbv/gerbv #94: Excellon format autodetection with leading zeros

neinseg, 2022-02-06:
The test file from #93 also confuses gerbv's Excellon format autodetection. All coordinates in the file have several leading zeros, so it should be obvious that trailing zero omission was used, yet, gerbv thinks this is a file with leading zero omission.

eyal0, 2022-11-28:
gerbv has autodetection?  I didn't even know this!

ooxi, 2022-11-28:
@neinseg does PR #156 fix this issue for you?

eyal0, 2022-11-28:
Don't merge it yet because I saw that some other test was affected and I can't tell if this caused a breakage elsewhere.

eyal0, 2022-11-28:
Ah already merged. Well, it's probably okay. I think what I saw what the handling of when there are two FILE_FORMAT lines in the same file. Do we honor the first or the second? Who knows? FILE_FORMAT is not even in the spec so it's hard to say what the right thing to do is. And this will probably never happen anyway.

ooxi, 2022-11-28:
@eyal0 PR #156 should not have been merged? I can revert the merge if you want, for the future I suggest we use Draft PRs