gerbv/gerbv #96: Incorrect rendering for Allegro Excellon files with "comment-style" tooldefs

neinseg, 2022-02-06:
gerbv does not parse the type of tool definition Allegro exports, and will render files using these tool definitions using some (incorrect) built-in defaults. Example file: [drill.zip](https://github.com/gerbv/gerbv/files/8009950/drill.zip)

Partial log:
```
** (process:857403): CRITICAL **: 11:35:53.848: Tool 01 used without being defined at line 22 in file "/home/jaseg/proj/gerbolyze/gerbonara/gerbonara/tests/resources/allegro-2/MinnowMax_RevA1_DRILL/MinnowMax_RevA1_NCDRILL.drl"
** (process:857403): WARNING **: 11:35:53.850: Setting a default size of 0.024"
** (process:857403): CRITICAL **: 11:35:53.851: Tool 02 used without being defined at line 1853 in file "/home/jaseg/proj/gerbolyze/gerbonara/gerbonara/tests/resources/allegro-2/MinnowMax_RevA1_DRILL/MinnowMax_RevA1_NCDRILL.drl"
** (process:857403): WARNING **: 11:35:53.851: Setting a default size of 0.032"
** (process:857403): CRITICAL **: 11:35:53.851: Tool 03 used without being defined at line 1877 in file "/home/jaseg/proj/gerbolyze/gerbonara/gerbonara/tests/resources/allegro-2/MinnowMax_RevA1_DRILL/MinnowMax_RevA1_NCDRILL.drl"
** (process:857403): WARNING **: 11:35:53.851: Setting a default size of 0.04"
** (process:857403): CRITICAL **: 11:35:53.851: Tool 04 used without being defined at line 1924 in file "/home/jaseg/proj/gerbolyze/gerbonara/gerbonara/tests/resources/allegro-2/MinnowMax_RevA1_DRILL/MinnowMax_RevA1_NCDRILL.drl"
** (process:857403): WARNING **: 11:35:53.851: Setting a default size of 0.048"
```

The syntax for these tool definitions is reasonably simple to parse, see [gerbonara's parser](https://gitlab.com/gerbolyze/gerbonara/-/blob/main/gerbonara/excellon.py#L578).

ooxi, 2022-02-27:
Supporting this feature, or at least warning the user if detecting this feature was used but not supported, would be great. I can offer guidance if you would like to propose a patch

eyal0, 2022-03-02:
They look like this in the file:
```
;T01 Holesize 1. = 8.000000 Tolerance = +3.000000/-3.000000 PLATED MILS Quantity = 1873
;T02 Holesize 2. = 36.000000 Tolerance = +3.000000/-3.000000 PLATED MILS Quantity = 23
;T03 Holesize 3. = 38.000000 Tolerance = +3.000000/-3.000000 PLATED MILS Quantity = 50
;T04 Holesize 4. = 40.000000 Tolerance = +3.000000/-3.000000 PLATED MILS Quantity = 7
;T05 Holesize 5. = 42.000000 Tolerance = +3.000000/-3.000000 PLATED MILS Quantity = 2
;T06 Holesize 6. = 62.000000 Tolerance = +3.000000/-3.000000 PLATED MILS Quantity = 2
;T07 Holesize 7. = 91.000000 Tolerance = +3.000000/-3.000000 PLATED MILS Quantity = 4
;T08 Holesize 8. = 125.000000 Tolerance = +3.000000/-3.000000 PLATED MILS Quantity = 2
;T09 Holesize 9. = 156.000000 Tolerance = +3.000000/-3.000000 PLATED MILS Quantity = 4
;T10 Holesize 10. = 20.000000 Tolerance = +5.000000/-5.000000 NON_PLATED MILS Quantity = 8
;T11 Holesize 11. = 31.000000 Tolerance = +5.000000/-5.000000 NON_PLATED MILS Quantity = 1
;T12 Holesize 12. = 43.000000 Tolerance = +5.000000/-5.000000 NON_PLATED MILS Quantity = 8
;T13 Holesize 13. = 46.000000 Tolerance = +5.000000/-5.000000 NON_PLATED MILS Quantity = 1
;T14 Holesize 14. = 118.000000 Tolerance = +5.000000/-5.000000 NON_PLATED MILS Quantity = 2
;T15 Holesize 15. = 125.000000 Tolerance = +5.000000/-5.000000 NON_PLATED MILS Quantity = 4
```

gerbv only supports the size and none of the other stuff so we'll ignore everything except size.  It's just the part after the = sign, right?

That parser that you linked to will be useful.  This won't be a small task but it should be doable.  We will use the function `drill_parse_t_code` as a reference and build off of it, just with different parsing.  Should be alright.

rampageservices, 2026-03-02:
@neinseg Would you be OK with the drill file from your original report (MinnowMax_RevA1_NCDRILL.drl inside drill.zip) being included in the gerbv test suite as a regression test for Allegro comment-style tool definitions?

The test would live alongside the other golden-reference tests under `test/inputs/` and `test/golden/`. If there are any licensing concerns with the MinnowMax board files, we can continue using a hand-crafted test file instead.