gerbv/gerbv #97: gerbv doesn't parse Altium Excellon number format comments

neinseg, 2022-02-06:
Here is an Excellon file that confuses gerbv's format auto-detection: 
[LimeSDR-QPCIe_1v2-RoundHoles.TXT](https://github.com/gerbv/gerbv/files/8009963/LimeSDR-QPCIe_1v2-RoundHoles.TXT)

This file is generated by Altium. Altium puts the number format of an excellon file inside a comment in the header like this:
```
;FILE_FORMAT=4:4
```

I think it would be worth it for gerbv to parse these comments and use them before guessing the number format.

ooxi, 2022-02-27:
I agree with your observation. This issue was the reason I got involved in gerbv initially :-)

eyal0, 2022-03-02:
How annoying!  Oh man, why can't everyone just follow the spec please?

I'll get to work.

eyal0, 2022-03-02:
Before #101 :

[image]

After:

[image]

Where did all the holes go?  Ah, have to zoom in:

[image]

Can someone confirm that it looks right?  Thanks.

closed 2022-03-29 by eyal0

ooxi, 2022-04-22:
Thank you very much for implementing this! I have bumped the version to `2.9.0-dev` since this is a really useful feature