gerbv/gerbv #98: Additional drill holes shown for a Zuken CADSTAR gerber output

ma-bx, 2022-03-01:
A PCB with 3 non-plated drill holes and 3 plated drill holes was generated by Zuken Cadstar 18.0 and imported into Gerbv 2.8.3-dev~82d59a
The gerber output also shows only 3 holes for each file.

The Autodetect feature was disabled with the drill files used, they were set to metric. 
Gerbv shows 5 plated and 5 non-plated drill holes.

[image]
[image]
 
DrillNonPlated.drl:
M48
FMAT,2
T1C4.0000
%
T1
X0050000 Y0250000
X0050000 Y0050000
X0250000 Y0250000
M30

DrillPlated.drl:
M48
FMAT,2
T1C2.2000
%
T1
X0100000 Y0200000
X0100000 Y0100000
X0200000 Y0200000
M30

eyal0, 2022-03-01:
gerbv is being confused by the space character in your files.

Instead of:

```
X0100000 Y0200000
```

It should be:

```
X0100000 Y0200000
```

I will check the spec of the Gerber language to see if:

1. The output made by Zuken Cadstar is invalid, or
2. The interpretation by gerbv is wrong.

eyal0, 2022-03-01:
It's not clear to me from the documentation which way it should be: https://gist.githubusercontent.com/katyo/5692b935abc085b1037e/raw/32879038a55e1e7019902d8d073d6f5d6d2e45ff/excellon.md

In all the examples, there are no spaces between X and Y.  However, there is no mention about whether or not spaces are allowed.  There's also no mention of newlines.

Can the Zuken Cadstar people be contacted?

ma-bx, 2022-03-01:
Thanks for checking! 

In Cadstar 18, there is a configuration file for the drill output files "Excellon.usr" where the number of spaces can be configured. Line "SPACES 1" seems to be the default. Changing it to "SPACES 0" as a workaround gets the desired results. 

However, it would be great to tolerate spaces if they are not prohibited by the Gerber spec. The Reference Viewer seems to tolerate spaces. If it is prohibited to use spaces, a warning or error would help if a user tries to load such files.

ma-bx, 2022-03-01:
> Which reference viewer did you use?

This one https://gerber-viewer.ucamco.com/, as mentioned on #77

eyal0, 2022-03-02:
@ma-bx If you're a programmer, perhaps you can review my code?  It should now support whitespace.  It also supports having more than one `X` or `Y` on the line, and it will just use the last one, which is similar to what kicad is doing.  Also, you can put `X` and `Y` in any order.

What do you think?  Looks okay?

ma-bx, 2022-03-02:
@eyal0 My programming skills are rudimentary, so I am not confident to properly review the code, sorry!

ma-bx, 2022-04-15:
I can confirm that it is solved in the latest version. Thanks a lot! 
[image]

PR #99 (2022-03-28): Support spaces in excellon XY drill lines
This fixes #98