gerbv/gerbv #106: Please bring back manual layer realignment feature

electroniceel, 2022-04-15:
Older versions of gerbv (<2.7.0) had a feature to manually realign layers. It was accessible by right-clicking on a layer name and then choosing "Modify orientation" in the context menu. This brought up a dialog which allowed you to shift and scale the current layer by arbitrary numbers.

You can see screenshots and a description of the feature for example here: https://docs.oshpark.com/design-tools/gerbv/modify-orientation/

Current versions (2.7.0 to 2.8.2) don't have this anymore. There is now a "Align layers" menu options in the Edit menu. But it doesn't always work (sometimes gives me "Can't align by this type of object") and I prefer to being able to manually tune the alignment anyways.

Thanks.

ooxi, 2022-04-16:
Unfortunately the screenshots on oshpark are of quite a low resolution, but they look similar to the *Edit Layer* dialog:

[image]

Is this what you are looking for?

electroniceel, 2022-04-16:
You are right, this is what I was looking for. Seems I didn't look closely enough.

Sorry to bother you.

closed 2022-04-16 by electroniceel