gerbv/gerbv #107: gerbv should be able to detect when a project file is passed as argument without option --project="..."

Dromantor, 2022-04-19:
From what I could see gerbv currently expects that each argument that is not an option is a valid gerber/drill file. The problem is that if you just configure to open .gvp with gerbv it fails to do so because your OS (I'm on Windows btw.) doesn't know about the needed --project="..." option. You currently have to manually modify the command line used to open your project file, which may or may not be easy depending on your OS.

Another problem with projects: The file `init.scm` that seems to be required for opening a project isn't packaged with the builds.

ooxi, 2022-04-22:
This does make sense from a usability point of view. However, the argument parsing code is quite elaborated. I would be happy to mentor such a patch however, if you want to create a PR.

Could you post an error message regarding `init.scm`? Indeed, it is not bundled and might be a simple fix.

Dromantor, 2022-04-22:
I'll try to come up with something when I got some spare time.

And regarding the `init.scm` error: I guess you already found the relevant error message yourself :)

ooxi, 2022-04-22:
@Dromantor the `init.scm` issue #111 was resolved by pull request #112 and will be included in the next release. You can use [snapshot builds](https://gerbv.github.io/#download) until then.

ooxi, 2023-07-16:
Fixed by PR #184

closed 2023-07-16 by ooxi