gerbv/gerbv #129: Is the BUGS file current?

spe-ciellt, 2022-07-04:
Hello!

I found the file `BUGS` in the root directory of the repository. It apparently lists a number of bugs. Are the bugs in the list still valid?
They are also built into the program when it is built. If they are valid, should they be built into the program? Seems more  like they should be here and then point to here in the "About" in the program.

ooxi, 2022-07-11:
@spe-ciellt I agree. If anything should be built into the binary, it should be a link to the GitHub issues, or a list of GitHub issues marked as verified bug

ooxi, 2022-07-12:
I thought about it some more and came to the following assesment

1. Do not include BUGS file in gerbv: Might break distributions using this file to include notes about their fork
2. Change contents of BUGS file to simply point to this issue tracker: easiest solution, but does not correctly reflect the bugs at the time of the binary creation
3. Manually gather BUGS contents: will be out of sync again in a short amount of time
4. Automatically gather BUGS contents from GitHub at time of build: requires network access during build
5. Automatically change BUGS contents using GitHub actions: Would be possible using [`on` `issues` `types` `labeled` / `unlabeled`](https://docs.github.com/en/actions/using-workflows/events-that-trigger-workflows#issues): Could cause too many commits and would be complicated to make sure verified bug lables are only placed by certain people

I therefore propose (2)

spe-ciellt, 2022-07-12:
Do anyone use the BUGS file for anything, like any distribution?

I would go with (1) and then update the "About..." screen to point at this repository. In my world this is where development on gerbv now happens and it would be great to point more people here. Else number (2) would suffice.

I guess 1 is more job than 2?

ooxi, 2022-07-12:
@spe-ciellt yes, (1) is more work than (2) but it's cleaner. I just don't want to step on maintainer's tows.

I will look at the main distributions packaging gerbv (Debian, Fedora, Ubuntu) to see, whether it's used somewhere.

PR #132 (2023-04-21): Update BUGS file to point to issue tracker
The BUGS file was no longer kept up to date since half a decade ago. Therefore it's more user friendly to point to the issue tracker.

I have not updated the translation files. Will fix #129
closed 2023-04-21 by ooxi