gerbv/gerbv #130: libgerbv.pc version doesn't match

eyal0, 2022-08-25:
https://github.com/pcb2gcode/pcb2gcode/issues/626#issuecomment-1226755727

Could this be caused by this new gerbv? The problem starts around the same time as this repo...

eyal0, 2022-08-29:
When I compile gerbv the usual way, a libgerbv.pc file is created:

To do it, type this:

```bash
./autogen.sh && ./configure --prefix=$HOME/.local && make -j 48
cat src/libgerbv.pc
```

In the output, we see the contents of a valid `libgerbv.pc` file with a valid version number in there.

I am using MX Linux version 21 on my computer.  I can download the debian install package that my computer will use and examine it like this:

```bash
sudo apt install --download-only gerbv
ls -l /var/cache/apt/archives/gerbv
cd ~/tmp
mkdir gerbv
cd gerbv
ar x gerbv*
tar xvf data.tar.*
cat `find . -name libgerbv.pc`
```

In the output, I see a valid `libgerbv.pc` file with a version number.

Now I try the same with the file avaialble from https://packages.ubuntu.com/jammy/gerbv

```bash
cd ~/tmp
mkdir gerbv2
cd gerbv2
wget 'http://mirrors.kernel.org/ubuntu/pool/universe/g/gerbv/gerbv_2.8.2-1_amd64.deb'
ar x gerbv_2.8.2-1_amd64.deb 
tar xvf data.tar.*
cat `find . -name libgerbv.pc`
```

I find no valid `Version` in libgerbv.pc.  This will cause configure scripts which I looking for gerbv of a specific version to fail, which is causing users to fail to compile pcb2gcode at in https://github.com/pcb2gcode/pcb2gcode/issues/626 .

It seems that the issue is due to how gerbv was packaged by the maintainers of Ubuntu jammy.  The site for that debian file is here: https://packages.ubuntu.com/jammy/gerbv

On that site, it says:

> Please consider filing a bug or asking a question via Launchpad before contacting the maintainer directly.

I will use the link to ask a question and see what the response is.

eyal0, 2022-08-29:
The question is here: https://answers.launchpad.net/ubuntu/+source/gerbv/+question/702967

eyal0, 2022-08-30:
Bug report: https://bugs.launchpad.net/ubuntu/+source/gerbv/+bug/1988117

stefanhusmann, 2022-09-13:
Same under Arch Linux based distros. The official package in [community] also lacks Version number in libgerbv.pc. Coincidence, that two packagers make the same mistake? I doubt.

eyal0, 2022-09-14:
No one is responding on the bug report. Maybe contact the guy directly?

stefanhusmann, 2022-09-14:
I think the problem is that the M4 magic is trying to derive the version number using "git decribe" or so. This fails if we build from the tarball.

eyal0, 2022-09-17:
Yes, I'm seeing this, too.  The source of the file is `./src/libgerbv.pc.in`.  I'll look into it.

closed 2022-09-18 by ooxi