gerbv/gerbv #147: ci Warnings

ooxi, 2022-11-04:
## ci-coverage

* Node.js 12 actions are deprecated. For more information see: https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/. Please update the following actions to use Node.js 16: coverallsapp/github-actionShow less
* The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show less
 

## ci

* Node.js 12 actions are deprecated. For more information see: https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/. Please update the following actions to use Node.js 16: JamesIves/github-pages-deploy-actionShow less
* The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/

ooxi, 2022-11-04:
github-pages-deploy-actions warnings fixed by pull request #148

eyal0, 2022-11-04:
There's a warning to be fixed in the coveralls app: https://github.com/coverallsapp/github-action/pull/117

The author of the app hasn't been active lately.  We could either just live with it or fork the repo and fix it ourselves.