gerbv/gerbv #149: aving settings after closing the program...

mikhailmihalkov, 2022-11-09:
After reopening the program, the latest settings are not saved... (View units )/( rendering) All reset to mil and Fast by default

ooxi, 2022-11-09:
Thanks for the report @mikhailmihalkov! Which version of gerbv are you using on which operating system?

mikhailmihalkov, 2022-11-09:
> Спасибо за отчет@mikhailmihalkov! Какую версию gerbv вы используете в какой операционной системе?

2.9.4 win 10/64

mikhailmihalkov, 2022-11-10:
oxxi There is a version for linux
https://launchpad.net/ubuntu/+source/gerbv/2.9.4-1/+build/24626945

ooxi, 2022-11-10:
It will take a bit to review this on my side

ooxi, 2022-11-20:
OK I can reproduce the issue. Looking into the source code, it does not seem like this feature ever existed.

@mikhailmihalkov to confirm, is this a regression report or a feature request?

mikhailmihalkov, 2022-11-20:
Эта функция работает на ubuntu но не работает на win

ooxi, 2022-11-20:
Thanks! Then I might have missed something, will look into it

mikhailmihalkov, 2022-11-20:
> Спасибо! Тогда я мог что-то упустить, посмотрю

When installing from the official repository, deb package

ooxi, 2022-11-21:
OK this might have something to do with issue #144. My current working hypothesis is that the schema cannot be found and thus storing / restoring of setting in Windows is disabled.

mikhailmihalkov, 2022-11-21:
In my understanding, the settings should be saved in the configuration file so it works on ubuntu.. After closing the program, the latest settings are saved and do not require re-entering during subsequent starts

ooxi, 2022-11-21:
@mikhailmihalkov yes, this is my understanding, too. And I suspect something going wrong while looking up the schema, thus disabling this functionality.

StephaneIFO, 2023-07-10:
Hi, 

I can confirm this problem on v2.9.8 under Windows10 H2
While waiting for this function to be fixed, is it possible to locally modify a file so that the default unit is mm and not mils?

thanks