gerbv/gerbv #154: Gerber file discovery win 10

mikhailmihalkov, 2022-11-21:
If the gerbv program is assigned by default, the gerber file cannot be opened (empty window)... File opens only after starting the program (menu open) win 10\64

ooxi, 2022-11-21:
@mikhailmihalkov which file extensions are you associating with gerbv? A gerbv project file or gerber layer files themselves?

ooxi, 2022-11-21:
Similar to issue #107

mikhailmihalkov, 2022-11-22:
> @mikhailmihalkovкакие расширения файлов вы связываете с gerbv? Файл проекта gerbv или сами файлы слоя gerber?

I assign to all types that generate cad and, when saved by the program itself, does not open gerbv in the same way.... gbr.

mikhailmihalkov, 2022-11-22:
> @mikhailmihalkovкакие файлы расширения вы связываете с gerbv? Файл проекта gerbv или сами файлы слоя gerber?

Yes, I make an appointment to open with....

mikhailmihalkov, 2022-11-22:
> Аналогично выпуску [№107](https://github.com/gerbv/gerbv/issues/107)

A similar problem is not solved... Did not see it when opening the request)

gyachts, 2023-01-04:
> Similar to issue #107

I would be interested in adding the feature to Windows builds to detect a project file (.gvp ext) so Windows can start gerbv with the project loaded without the command line switches of '--project=<prjctfile'.  I currently solved it using a Power Script.
Do you want me to make a new PR or use #107?