gerbv/gerbv #155: Numeric format incorrect when reading Altium 14 NC Drill file

meantaipan, 2022-11-27:
gerbv is misreading the numeric format in an Altium Excellon drill file.  The top of an example is here:

```
M48
;Layer_Color=9474304
;FILE_FORMAT=2:3
INCH,TZ
;TYPE=PLATED
T1F00S00C0.028
T2F00S00C0.035
T3F00S00C0.040
T4F00S00C0.047
T5F00S00C0.051
T6F00S00C0.072
T7F00S00C0.125
%
T01
X1875Y350
X1965Y460
etc.
```

Tracing execution in drill.c, it is correctly parsing the ```FILE_FORMAT=2:3``` line, but the subsequent ```INCH,TZ``` line resets to trailing zero suppression with 4 decimals.  The file then reads at 1/10 of the proper scale.

I don't know if Altium is at fault, but without a detailed knowledge of the standard, it looks incorrect to set TZ because the rest of file is obviously not suppressing trailing zeros (a dumb concept, but there you have it).  But an Excellon guru needs to look at it.

If, in gerbv viewer, the format is manually reset to 3 decimals (no autodetect), it reads at correct scale.  

Possibly, ```state->autod``` should be reset if the ```FILE_FORMAT``` line is read, or another flag should be set to prevent any change to the format.

This may be related to issue #94.

eyal0, 2022-11-28:
I worked on stuff involving the parsing:

https://github.com/gerbv/gerbv/commit/6ab3d2203fcea6b3f3216bc56253f249668c604b
https://github.com/gerbv/gerbv/commit/4f3b06c58c616dfc6b9e66111b6a6063a1a2d6f7
https://github.com/gerbv/gerbv/issues/97

Those were about Altium, too!

Let's start with the spec and see how it's *supposed* to work:

https://www.ucamco.com/files/downloads/file_en/416/the-gerber-layer-format-specification-revision-2021-02_en.pdf?1109f5f3efc7ef822119bd893d685572

eyal0, 2022-11-28:
I don't see it in there at all!  Ah, damn, that's the gerber format this is a drill file.  Grr.  Okay, Excellon spec.

I see the behavior that you are talking about.  If I open the lines that you provided:

[image]

We see that the location is 0.1965, 0.046 in inches.  The diameter of the hole seems fine.

If I swap the `FILE_FORMAT` and the `INCH,TZ` line then I get this:

[image]

Now we see that the location ix 1.875,0.35.  That's more correct.  Again, diameter is fine.  So it seems like the image is being scaled as described but the holes are staying the same size.

meantaipan, 2022-11-28:
Yes, hole sizes are correct, probably because the T..C.. lines give the hole size as a floating point with an explicit decimal point.  So the parsing is distinct from the positioning lines, which rely on the implicit radix point position.

Note that I tried the trick of making state->autod into a bitfield.  Initially set to 3 (bits 1,0) but if the FILE_FORMAT line is seen, then it turns off bit 0.  Thereafter, the decimals value is only changed if bit 0 is still set (e.g. if the INCH,TZ line is read).  The other bit (bit 1) enables all the other autodetect functionality.  This fixes my particular issue.

Another data point: my board house (Advanced Circuits) read the gerbers and NC drill files no problem, and I got good boards.  If you wish I can donate the complete set of files for testing.

eyal0, 2022-11-28:
What if we just set `state->autod` to false after we see the `;FILE_FORMAT` comment?  The handling of formats is confusing to me but it seems like it would work...  It's working in my test.

meantaipan, 2022-11-28:
That was my first thought, but I don't have the Excellon spec available (paywalled) so was not sure whether that would be a proper fix.  But if it works for you, then it is a more straightforward fix than mine.

<rant>It's a shame that we have to worry about this.  It would have been a lot easier to just insist on decimal points throughout (like milling machine or lathe G-code), and what genius thought that trailing zero suppression would ever be a good idea!?  But it is what it is...</rant>

eyal0, 2022-11-28:
Back when Excellon was invented I guess that people thought that it would be more important to save the space in the file and remove decimal points?  Who knows?  I'll have a fix ready soon.

eyal0, 2022-11-28:
This guide seems pretty good: https://web.archive.org/web/20071030075236/http://www.excellon.com/manuals/program.htm

meantaipan, 2022-11-28:
Just been reading that spec.  Unfortunately, it doesn't mention the "hot comments" like ```;FILE_FORMAT``` that seem to have been added by other vendors etc.  So it's still a bit confusing about what takes priority and whether order is important.

I'll give your fix a try...

ooxi, 2022-11-28:
@meantaipan PR #156 will be included in release 2.9.6

Feel free to reopen this issue if it does not solve your problem.

PR #156 (2022-11-28): Turn off autodetect after FILE_FORMAT
This fixes #155
closed 2022-11-28 by ooxi