gerbv/gerbv #157: Enhancement request: generalize drawing.

meantaipan, 2022-11-28:
Currently, the code basically relies on Cairo for rendering (pixmap or vector format).  I would like to be able to generate output suitable for use with OpenSceneGraph and similar 3D rendering libraries.  This would generate board views similar to what most EDA tools can do, that can be flipped around with the mouse with realistic color, thickness etc.

I am prepared to do this work, but I am asking here whether this is the best place/repository to do it.  Thoughts?

The approach I have in mind at present is to change direct calls to Cairo (e.g. ```cairo_move_to()```) to indirect calls via function pointers: ```draw_impl->move_to()```.  Instead of passing ```cairo_t *cairoTarget``` around, pass a struct that is one additional level of abstraction away from Cairo.  Naturally, a default object would be set up to use Cairo, so that library functionality is unchanged by default, however a library user could create their own draw implementation which uses OSG instead of Cairo, provide they stick to Cairo-compatible semantics.

Another enhancement is to support IPC-D-356A test data (I already use this to run a soldering robot but would be nice to integrate graphically), and this in turn suggests support for RS274-X2 (attributes) as a way to store the data.

Open to suggestions...

eyal0, 2022-11-28:
`pcb2gcode` uses gerbv as just a parser of the file and converts the entirety of a gerber file into a ["well known text"](https://en.wikipedia.org/wiki/Well-known_text_representation_of_geometry), which is a common way of representing lines and shapes and used a lot for GIS stuff (like maps).  WKT is easily converted to SVG files, they are very similar in structure.

The entrypoint for the conversion is here:

https://github.com/pcb2gcode/pcb2gcode/blob/master/gerberimporter.cpp#L735-L738

The `render` function needs to know if you want to fill closed lines (this is useful for things that are drawn as an outline, such as edge cuts layers).  You can choose to either return just shapes or shapes and lines, where the lines are for traces that were cut out with a circular tool.

The tool will use libgeos if it is installed and if not, it will use boost.  Either way, boost must be installed.

The input is the gerber file opened by libgerbv.

If you check out the repo, run `autoconf -fvi && ./configure && make -j 20 check`.  Then I suggest that you look at the `gerberimporter_tests` because it is testing *just* the gerberimporter.

Try adding this line to the top of the file `gerberimporter_tests.cpp`:

```c++
#include <boost/geometry/io/wkt/write.hpp>
```

And then add this on line 205, right after the render:

```c++
  std::cout << bg::wkt(polys) << std::endl;
```

You'll get outputs like this:
MULTIPOLYGON(((3.0226 2.63891,3.11665 2.60495,2.7771 1.66437,2.68304 1.69832,3.0226 2.63891)),((0.987563 -0.0300261,0.981944 -0.0270228,0.977019 -0.022981,0.972977 -0.018056,0.969974 -0.0124372,0.968124 -0.00634044,0.9675 0,0.9675 5,0.968124 5.00634,0.969974 5.01244,0.972977 5.01806,0.977019 5.02298,0.981944 5.02702,0.987563 5.03003,0.99366 5.03188,1 5.0325,6 5.0325,6.00634 5.03188,6.01244 5.03003,6.01806 5.02702,6.02298 5.02298,6.02702 5.01806,6.03003 5.01244,6.03188 5.00634,6.0325 5,6.0325 0,6.03188 -0.00634044,6.03003 -0.0124372,6.02702 -0.018056,6.02298 -0.022981,6.01806 -0.0270228,6.01244 -0.0300261,6.00634 -0.0318755,6 -0.0325,1 -0.0325,0.99366 -0.0318755,0.987563 -0.0300261),(5.9675 4.9675,1.0325 4.9675,1.0325 0.0325,5.9675 0.0325,5.9675 4.9675)),((2.85208 4.60504,4.21032 0.8427,3.88111 0.723854,2.52288 4.48619,2.85208 4.60504)))

That's a multipolygon, which means that is has multiple polygons inside of it.  If you then copy and paste that text into this website https://jsfiddle.net/eyal0/z4px7g30/12/ you will see the shape.  (Don't forget to click "run" at the top.)

Perhaps OSG already knows how to deal with well-known-text or SVG?

Yes, you've got some work ahead of you!

eyal0, 2022-11-28:
I never needed to separate it into a library so I didn't do that.  But it ought to be possible.  If you did that work, I would incorporate it here, sure.

I don't know how to combine all the netlist info together with the gerber.  If there's a format for it then it probably wouldn't be too much work to modify the gerberimporter to do it.  All those boost classes that are used in the gerberimporter like point, line, multiline, polygon, etc, are pretty easy to work work.  A point is just a pair of numbers, a line is a vector of points, a multiline is a vector of lines, etc.  Casting them to a different type so that you can attach more data is a breeze.

eyal0, 2022-11-28:
Also, WKT is kind of a standard so it's easy to work with.  For python, you wouldn't need to parse it yourself, just use a library I think.

https://pypi.org/project/wktplot/

meantaipan, 2022-11-28:
To be clear, I think it would be possible to add netlist (and other attribute) data to gerbv, without breaking the existing API.  As far as I know, the API does all the structure allocation so if extra stuff is tacked on it should not be a problem.

There's probably a smarter way to do this, but to incorporate 356A data which basically associates coordinates with component pin and netlist data, I think it would be sufficient to use a "point in shape" test which should be reasonably efficient given the simplicity of most pad shapes.  It would need a bit more work to find via connections in multi-layer boards, but similar in principle.  Finding point membership in general polygon shapes is a bit more complex, but simple enough to implement without requiring boost.  You've already got that point test for highlighting etc.

356A format is very simple fixed fields.  I wrote an 80-line Python module to parse them looking for nets and pins.  Here's a snippet from one of my projects:

```
C  Project Name : 
C  Board Name : 
C  Date : 2/1/2021
C  Time : 2:43:22 PM
C  
C  --------------- Layer Stackup --------------------
C  
C  Top Layer
C  Bottom Layer
C  
C  --------------------------------------------------
C  
P  JOB
P  UNITS CUST
P  TITLE ProC4C4.tmp
P  NUM 1.0
P  REV A
P  VER IPC-D-356A
P  IMAGE PRIMARY
317INB              P6    -19   D0394PA00X 022750Y 013000X0000          S0
317INB              J1    -2    D0429PA00X 042941Y 006909X0000          S0
317INB              P12   -2    D0354PA00X 043484Y 014700X0000          S0
317INC              P2    -10   D0354PA00X 041250Y 010250X0000          S0
317INC              J1    -9    D0429PA00X 044059Y 009636X0000          S0
317INC              P12   -1    D0354PA00X 042500Y 014700X0000          S0
327NetJ1_3          U2    -5         PA01X 035616Y 007108X0138Y0650R000 S0
327NetJ1_3          U2    -6         PA01X 035872Y 007108X0138Y0650R000 S0
327NetJ1_3          U1    -3         PA01X 037250Y 007988X0335Y0157R000 S0
317NetJ1_3          J1    -3    D0429PA00X 042941Y 008000X0000          S0
317NetJ1_3          P1    -2    D0354PA00X 039000Y 008250X0000          S0
317NetJ1_3          P2    -5    D0354PA00X 040250Y 008250X0000          S0
327NetJ1_4          U1    -1         PA01X 037250Y 008500X0335Y0157R000 S0
327NetJ1_4          U2    -3         PA01X 035872Y 009392X0138Y0650R000 S0
327NetJ1_4          U2    -2         PA01X 036128Y 009392X0138Y0650R000 S0
317NetJ1_4          J1    -4    D0429PA00X 042941Y 009091X0000          S0
317NetJ1_4          P1    -3    D0354PA00X 039000Y 009250X0000          S0
317NetJ1_4          P2    -7    D0354PA00X 040250Y 009250X0000          S0

```

ooxi, 2022-11-28:
libgerbv is a great library to parse Gerber and transform it to a different format. For converting to 356A I would suggest using libgerbv API instead of substituting the Cairo/GTK renderer.

If the tool is small enough, including it into gerbv might be viable.

meantaipan, 2022-11-28:
Agreed, I am no longer proposing changing the rendering part.  I am proposing to allow 356A as another input file to libgerbv, which will then annotate existing layers and other structures with designator, netlist etc. info.  As mentioned, I think this is feasible without huge amounts of code or changing the current API, and only appending to current data structures.

ooxi, 2022-11-28:
@meantaipan I'm open to extending the PI to support 356A if you would like to give it a try. While I do not have the resources to do the implementation, I'm more than happy helping you navigate gerbv's source code if you have any questions

meantaipan, 2022-11-28:
That's great.  Yes, I mentioned above that I was going to do the work.  I just didn't want to start without getting some buy-in from existing developers.

I probably will have some questions.  If this thread is appropriate for that then I will continue to use it.  Otherwise, I'm open to other suggestions.  I'm an experienced code developer, but bit of a novice with github etiquette.

ooxi, 2022-11-28:
@meantaipan feel free to use this thread for discussion. GitHub also provides a [bulletin board like subsystem](https://github.com/gerbv/gerbv/discussions) but it's not enabled for this project since it would duplicate the mailing list.

meantaipan, 2022-11-30:
First stage is almost complete, which implements RS274-X2 attributes (```%TF...*%``` etc.).  The library can now read and write such files.  The exporter is also a bit cleaned up: if there are any file attributes, then it writes X2 according to the most recent recommendations, otherwise it is the same as before.  Next stage is to implement the IPC-D-356A reader and annotations of images via the attribute facility.

Do you guys want to see development milestones like this?  I can push it up to my fork.  I think it would be helpful if you can give a bit of a code review to make sure I'm not doing things the wrong way.

If I do multiple commits (little bug fixes etc.) do those commits show in the history when you pull my changes?  Or can you squash out all that stuff?  Sorry, my git fu is very basic.

mikhailmihalkov, 2022-12-01:
What is really missing is the ability to edit the pads in the gerber (macro form view)

meantaipan, 2022-12-01:
Edits like that are a bit outside the scope of this enhancement, unless I misunderstand.  This is really about annotating pads and tracks etc. with netlist data, component designators, pins and so on.  When this data is available, then it will be easier to write a smart utility that could (say) change all pads of U15 to a different size.

Anyway, for those who are interested I have pushed up a commit to meantaipan/gerbv.  Known issue is that %TD (attribute deletions) may cause slight semantic differences between input and exported gerbers.  Fix in the pipeline.

mikhailmihalkov, 2022-12-01:
In gerbv, I see this data but I can't change it... The maximum is to delete the entire part

ooxi, 2022-12-01:
> First stage is almost complete, which implements RS274-X2 attributes (%TF...*% etc.). The library can now read and write such files. The exporter is also a bit cleaned up: if there are any file [...]

that's great news!

> Do you guys want to see development milestones like this? I can push it up to my fork. I think it would be helpful if you can give a bit of a code review to make sure I'm not doing things the wrong way.

I for sure would be interested! No pressure however :-)

> If I do multiple commits (little bug fixes etc.) do those commits show in the history when you pull my changes? Or can you squash out all that stuff? Sorry, my git fu is very basic

Unless the commits are specially crafted to tell a story (e.g. preparing internal subsystems before adding a feature) I would prefer to squash. Could you estimate the size of the changes?

mikhailmihalkov, 2022-12-01:
ooxi In fact, it is enough just to change the value of the description from a complex form to primitives and delete macros.... For example, an oval on a rectangle is necessary when editing the herber file of a stencil.... At the moment, it is possible only to delete completely... The second trouble is that there is no such software under linux at all.... Even kicad does not know how to disable the macro as much as possible

meantaipan, 2022-12-01:
@ooxi the changes so far are two new source files, a new subdirectory in example for test cases.  Most modifications to existing code are in export-rs274x.c.  Fairly small changes in a sprinkling of other files.  I'll try to reserve commits for milestones.

BTW I am using "x2attr" in most cases to denote this type of attribute, to avoid confusion with existing uses (HID attributes).

At present I am otherwise engaged, but in the meantime you can see the changes in my fork.

meantaipan, 2022-12-08:
New commit in my fork.  Implemented IPC-D-356A file reader, and commandline options to specify some processing options, particularly to help break up the data into distinct layers (images).  Here's a screenshot of the gerbv utility.  Look mom, no Gerbers!  The images were all generated from the IPC data alone.  The data does not contain complete thru-hole pad sizes, so these are shown with thin annular sections.  Next phase is to implement annotation of the original gerbers with netlist and component data.

[image]

meantaipan, 2022-12-14:
Hey guys, I am running the regression tests, and have come across a difference...

```
Test:  example_ekf2_l0
../src/run_gerbv -- --export=png --window=640x480 --output=outputs/example_ekf2_l0.png  ./inputs/../../example/ekf2/l0.grb
Set LD_LIBRARY_PATH to /home/steve/gerbv/src/.libs
Set LD_RUN_PATH to /home/steve/gerbv/src/.libs
Set LD_PRELOAD to /home/steve/gerbv/src/.libs/libgerbv.so
FAILED:  See mismatch/example_ekf2_l0
```

The diff image is a faint rendering of the board, with a few small red highlights:

[image]

The above shows one of them.  Other is similar.  Seems to be associated with square pads.  Normal rendering seems OK to me, for this test case.

What do you suggest?  I don't want to regenerate golden just so it works on my system.  Is there a way of bypassing individual tests so I can get to the end?

eyal0, 2022-12-15:
If your goal is to get through the tests, just `--regen` the one that is broken.

But the better question is *why* is the test failing?  Did your code change something that would cause this?

I suggest that you do a regen of all the "golden" files by running `./run_tests.sh --regen` in the `test` directory.  Then `git add -A` and `git push` to the PR.  Then we can all view all the diffs and see what's going on.

We can have this discussion in the comments of the PR.

meantaipan, 2022-12-15:
I checked this particular case, and it seems to be nothing to do with my changes, but simply because that is a very detailed example Gerber but is being rendered into 640x480 pix.  Most of the pads  and vias are getting obliterated, so it is a matter of rounding error and phase of the moon whether the square pads are getting displayed.  We are talking single pixels here.  Also the test script doesn't work quite properly on my system.  It's bailing out after first error because of the set -e at the top.  It would be nice if there was a --soldier-on option.

Apart from single pix errors, all tests are working except I seem to have broken export of text in the PnP example.  Probably because I made the text size dynamic depending on zoom level.  Not so sure you should even be pixel-wise comparing output text.  It's going to differ over systems based on different default fonts etc.  I'll fix my code so it writes text glyphs to exports, but don't expect diffs to pass.  

I am on Ubuntu 16.04 which is very old, so not surprising there are some insignificant rendering diffs.

I've added some test cases with IPC-D-356A data so I'll be adding them to the list.  Since this does not change the rendering, with default options, I think the test will be to export annotated Gerbers, then read them back in, export a PNG image, and compare that to the original.  I've been using that test myself, reading and writing several iterations then using diff on the files - they should be *exactly* the same after first iteration.