gerbv/gerbv #158: Enhancement request: more flexible Pick and Place input reader.

meantaipan, 2022-12-02:
The current PnP CSV reader seems to be specific to a particular column format.  Since CSV input is notoriously non-standard and hard to parse reliably, suggest deprecating the current PnP input reader and replace with ability to call an external program.  This means that the detailed implementation of a CSV reader is not part of gerbv, which will simplify maintenance.

To process PnP input, gerbv command line would have another option:

```
gerbv --pnp-reader="my-converter-script.py -extra -args"
```

```my-converter-script.py``` (or whatever) would be a translator program that converts the PnP CSV file into a standard canonical format that is easy for gerbv to parse.  If no ```--pnp-reader``` option is specified, then it uses the current reader for backwards compatibility.

Converter program would be called via system(3) or similar, with input and output file names as args, plus extra args:

```
my-converter-script.py  -extra -args  "<infile.csv>" tmp/gerbv-std-pnp.bin 
```

Something like that, anyway.

mikhailmihalkov, 2022-12-02:
Excuse me, why convert it? This file is usually loaded into the component installer program and is already being edited there....

meantaipan, 2022-12-02:
@mikhailmihalkov It's nice to be able to view component placement on top of the Gerber images.  This converter program is used so that random format PnP files can be viewed in gerbv.  It is already in there, but either make it work or take it out completely.

KjellMorgenstern, 2025-03-20:
I did some work on this here: https://github.com/KjellMorgenstern/gerbv/tree/refactor_pnp

Still work in progress, but it it works with the examples from gerbv (protel-pnp and pick-and-place subdir) and fritzings .xy export.

It tries to find a line that looks like a header, e.g.
`# RefDes, Description, Value, X, Y, rotation, top/bottom`
or
`RefDes,Description,Package,X,Y,Rotation,Side,Mount`

further tries to guess if SMD sizes are in Description, Package or Footprint column.