gerbv/gerbv #159: Flashes with holes draw artifacts when labels also drawn

meantaipan, 2022-12-10:
If labels are used on a flashed pad etc., and the pad has a drill hole defined, then there are some visual artifacts
drawn when rendering in normal or high quality, particularly when also rendering labels.

A simple fix is to modify the hole path drawing as follows:

```
static void
gerbv_draw_aperture_hole(cairo_t *cairoTarget,
		gdouble dimensionX, gdouble dimensionY, gboolean pixelOutput)
{
	if (dimensionX) {
	        cairo_new_sub_path(cairoTarget); // <------ SJH - added to clean up annular ring drawing
		if (dimensionY)
			gerbv_draw_rectangle (cairoTarget,
					dimensionX, dimensionY, pixelOutput);
		else
			gerbv_draw_circle (cairoTarget, dimensionX);
	}

	return;
}
```

This addition gets rid of the implied path segment between the outer and inner arcs, and cleans up the drawing.

It's probably a good idea to add `cairo_new_sub_path()` before label rendering too, but I haven't tried this.

eyal0, 2022-12-11:
Do you want to make this a PR so that we can run CI on it and see if everything else stays consistent?

Also, for this fix, we may need to add a test. If you have a sample file, we can use it. I can help with that.

meantaipan, 2022-12-11:
I only noticed this with my IPC-D-356A development work.  I'm right in the middle of that at the moment, so it will be awkward to do a PR.  Also, you would need all my other changes since the only other way to generate text labels at present is to use pick and place files, but I don't have any of those in the proper format (see #158).  Text in the rendering is the only way I know to generate the problem.

eyal0, 2022-12-11:
Okay, we'll wait. Does GitHub let us mark this issue as dependent on that issue?

Depends on #157