gerbv/gerbv #162: SEGV on export file as RS274-X

meantaipan, 2022-12-18:
The following valgrind trace was observed (relative to PR #161 ):

```
steve@charon:~/gerbv/test$ valgrind --trace-children=yes --suppressions=gerbv.supp --error-exitcode=127 --errors-for-leak-kinds=definite --leak-check=full -s --exit-on-first-error=yes --expensive-definedness-checks=yes --keep-stacktraces=alloc-and-free --  gerbv --export=png --window=640x480 --export=rs274x --output=outputs2/example_numpres_numpres.pcb.output_unplated-drill.grb-again outputs2/example_numpres_numpres.pcb.output_unplated-drill.grb
==27813== Memcheck, a memory error detector
==27813== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==27813== Using Valgrind-3.20.0 and LibVEX; rerun with -h for copyright info
==27813== Command: gerbv --export=png --window=640x480 --export=rs274x --output=outputs2/example_numpres_numpres.pcb.output_unplated-drill.grb-again outputs2/example_numpres_numpres.pcb.output_unplated-drill.grb
==27813== 
==27813== Invalid read of size 8
==27813==    at 0x4E5C227: gerbv_destroy_image (gerb_image.c:192)
==27813==    by 0x429943: main (main.c:1196)
==27813==  Address 0xbe73af0 is 48 bytes inside a block of size 56 free'd
==27813==    at 0x4C31740: free (vg_replace_malloc.c:884)
==27813==    by 0x4E5C23A: gerbv_destroy_image (gerb_image.c:193)
==27813==    by 0x4E5AB68: _export (export-rs274x.c:556)
==27813==    by 0x4E5ABFA: gerbv_export_rs274x_file_from_image (export-rs274x.c:577)
==27813==    by 0x4298D9: main (main.c:1181)
==27813==  Block was alloc'd at
==27813==    at 0x4C33914: calloc (vg_replace_malloc.c:1340)
==27813==    by 0x63D68E0: g_malloc0 (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==27813==    by 0x4E5C7AE: gerbv_image_return_new_netstate (gerb_image.c:338)
==27813==    by 0x4E6497E: parse_rs274x (gerber.c:1452)
==27813==    by 0x4E61115: gerber_parse_file_segment (gerber.c:268)
==27813==    by 0x4E62C05: parse_gerb (gerber.c:795)
==27813==    by 0x4E6AA8A: gerbv_open_image (gerbv.c:822)
==27813==    by 0x4E691FC: gerbv_open_layer_from_filename_with_color (gerbv.c:254)
==27813==    by 0x428AE9: main (main.c:966)
==27813== 
==27813== 
==27813== Exit program on first error (--exit-on-first-error=yes)
```

Problem seems to be in duplicating the image, net and layer state info is copied including the 'next' pointer, which is potentially dangling when the original image is freed, then the duplicate is freed.

Seems to be fixed with the following changes in `gerb_image.c`.

```
gerbv_layer_t *
gerbv_image_duplicate_layer (gerbv_layer_t *oldLayer) {
    gerbv_layer_t *newLayer = g_new (gerbv_layer_t,1);
    
    *newLayer = *oldLayer;
    newLayer->name = g_strdup (oldLayer->name);
	newLayer->next = NULL;  // SJH fix potential double free
    return newLayer;
}

static gerbv_netstate_t *
gerbv_image_duplicate_state (gerbv_netstate_t *oldState)
{
	gerbv_netstate_t *newState = g_new (gerbv_netstate_t, 1);

	*newState = *oldState;
	newState->next = NULL;  // SJH fix potential double free
	return newState;
}
```

ooxi, 2022-12-18:
While I agree with the issue, I'm unsure about the fix since it changes the semantics. Many calling duplicate recursive?

meantaipan, 2022-12-18:
@ooxi I did run all the regression tests, and nothing is changed except to fix the SEGV.  But what are the semantics?  I thought that the netstate is kept in a linked list merely for efficiency.  It seems to me that the process of duplication should not be leaving pointers to any data owned by another structure.  The contents of the `next` pointer at the point of duplication should be irrelevant to the owning structure, the `gerbv_net_t`.  It's possible this is generating too many copies rather than sharing a reference, so I'll check that.

meantaipan, 2022-12-19:
Yes, it is making too many copies.  It is making a `gerbv_layer_t` instance for each net, instead of referencing the one instance multiple times.  I think it's because of a pointer comparison rather than comparing the contents.  I'll see if I can fix it.

ooxi, 2022-12-19:
Feel free creating a PR and we'll merge it :-)

meantaipan, 2022-12-19:
Will do.  I am making some other fixes to RS274-X export, which are related to more extensive testing of the X2 features, so all that will be in there.  This time, all the tests are going to run fine :-)

meantaipan, 2022-12-19:
The problems I am finding are being found by the extended testing, so it would not really make sense to separate the testing script enhancement from fixing the problem which prevents test completion.  Don't worry, I try to keep fixes as orthogonal as possible.