gerbv/gerbv #165: SVG export creates white shapes instead of cutout

Riesi, 2023-01-23:
Exporting an SVG file from the following Gerber results in the proper shape, but some parts that should be cutout to the background are white shapes instead.

Gerber file:
[QMKMouse_Copper_Signal_Top.gbr.txt](https://github.com/gerbv/gerbv/files/10478919/QMKMouse_Copper_Signal_Top.gbr.txt)

Broken SVG:
[QMKMouse_Copper_Signal_Top.svg](https://user-images.githubusercontent.com/27691485/214021167-0b35ecd5-8bdf-45fd-a99b-e7dc619922d7.svg)

Riesi, 2023-01-23:
The shape looks correct in the GUI viewer, so maybe Cairo is at fault?
PNG export looks fine.

eyal0, 2023-01-24:
It's easier to see the white shapes when the background is not white:

[image]

eyal0, 2023-01-24:
It has to do with how the gerber file is and how gerbv is doing it.

The gerber file has purple areas filled and then the white areas as subtraction, to remove from the purple.  But gerbv is rendering each one separately so it is sort of drawing the white on top of the purple, as if it were a subtraction.

A better approach would be to have the white actually subtract from the purple shape, so that the purple shape has a hole in it.  That would require a library that can manipulate shapes so that a hole is created in the underlying shape.  gerbv doesn't have that so I guess that it's just drawing white.  :-(

In pcb2gcode I have my own svg rendering and it does the actual subtraction, using either boost geometry or libgeos to do the work.  I imagine that it would be difficult to put either of those into gerbv.

Depending on how important this is to you, I could show you how to do it in pcb2gcode.  How important is this to you?

Riesi, 2023-01-24:
Its not really important. I just wanted to report it, because it seemed like a valid issue.

I am using Gerbv to validate my KiCAD port of an Altium design (this Gerber) and at first I wanted to export the diffs. Now I just call Gerbv in a shell script on the multiple layers and look at that.

Thank you for the explanation though!

eyal0, 2023-01-28:
https://stackoverflow.com/questions/22579508/subtract-one-circle-from-another-in-svg

Maybe something like that could work?

Or perhaps post process the shape through something to convert white shapes to masks.