gerbv/gerbv #166: Bottom side view

htot, 2023-01-27:
It would be very useful if there would be "bottom side view" (vertical mirrored view) . This would allow to more easily compare the bottom layer with a real PCB.

mikhailmihalkov, 2023-02-27:
So in the settings there is a rotation of file or what are you talking about?

alexustas93, 2023-02-27:
Today for mirroring the bottom side you need go to the "Edit layer" and check the "Mirroring aout Y axis".
It's be more helpfull if the mirroring will be set on the toolbar and get the fast key (ex. "ctrl-M").

Also today the mirroring is not around the center of the board drawing. 
It's will be better if you will mirroring around the center of the board.

htot, 2023-02-27:
@alexustas93 I see. I did not notice that as I just wanted to change the view, not edit the layer. But now I see there are buttons for "Apply to active" and "Apply to visible". Very useful to know.

So, yes what I was looking for would be "Apply to all" and "ctrl-M" as a shortcut would be very nice. Actually without changing coordinates (looking from the other side of an object does not change the coordinates of the object itself). And looking from top side we see a stack of layers top-down. Bottom side view would then be bottom-up.

Nevertheless "Apply to visible" does the trick nicely (and in some cases better). I like the red M icon too.

ooxi, 2023-04-21:
@htot It seems that the shortcuts provided by @alexustas93 resolved your issue? Therefore I'm closing this issue.

Feel free to reopen if you would like further assistance!

closed 2023-04-21 by ooxi