gerbv/gerbv #167: Gerbv incorrectly rounds aperture coordinates in fast render mode

jaseg, 2023-04-05:
In fast render mode, gerbv currently seemingly rounds the screen space coordinates of rendered apertures to the pixel grid based on the top, left corner of the aperture's bounding box. This leads to artifacts where several apertures flashed at the same location are not always rendered concentric with each other when their bounding boxes are different. Ideally, gerbv should render 

An example where this rounding error is especially visible is a standard THT pad at low zoom as shown in the example below. At high zoom, the holes are correctly rendered centered on the pads. At low zoom, instead, the holes are rendered shifted to the top, left corner of the pad's bounding box.

Example screenshot at low zoom with the holes rendered shifted:
[image]

Example screenshot at high zoom with the holes rendered concentric:
[image]

gerbv version: Version 2.9.5 Compiled on Nov 29 2022 at 23:22:25, from the Archlinux package repository.

ooxi, 2023-04-21:
I have noticed something similar myself! Would you be able to attache the displayed gerber files?

While I am unfortunately not able to dig into this myself, I would be more than happy to help you out with a PR

jaseg, 2023-04-30:
@eyal0 here you go:
[gerbers.zip](https://github.com/gerbv/gerbv/files/11361279/gerbers.1.zip)
Magnified screenshot at low zoom level:
[image]
Also note that the arcs and outline on the bottom seem to show a similar issue, in the file there is no step there.

I generated the file using https://dyna.kokoroyukuma.de/protogen/#

ooxi, 2023-05-26:
@jaseg PR#173 by @eyal0 was merged as 9d53242a141aae8722c1d12aca9d07b2f44e2a04. Would be great if you could give it a try and reopen this issue / create another issue if there is feedback.

jaseg, 2023-06-03:
Subjectively I'd say it's better than it used to be, but it still happens. I just tested a fresh build from main, and this is what I got for gerbers.zip from my previous comment. Note that the outline looks much better now, but the holes are still wonky. Since this issue is about apertures, not arcs, I think reopening it would make sense - I don't have the github permissions for that though.

[image]

PR #173 (2023-05-25): fix: Using floating point math for arcs.
Arcs and lines don't always line up when the image is zoomed out because gdk expects integers so we do a lot of integer math.  For example, under a lot of zoom out, we might be looking at arcs circles with a diameter of 5.  When dividing by 2 to compute the radius, this causes a problem.

The solution is to do as much floating-point math as possible and only convert to integer at the very end, when we call the GDK functions to draw.  In testing, this fixed #167.
commit 9d53242a (2023-05-25): fix: Using floating point math for arcs.

Arcs and lines don't always line up when the image is zoomed out
because gdk expects integers so we do a lot of integer math.  For
example, under a lot of zoom out, we might be looking at arcs circles
with a diameter of 5.  When dividing by 2 to compute the radius, this
causes a problem.

The solution is to do as much floating-point math as possible and only
convert to integer at the very end, when we call the GDK functions to
draw.  In testing, this fixed #167.