gerbv/gerbv #168: File type of empty XNC file not detected correctly

jaseg, 2023-04-05:
When loading the following empty XNC file, gerbv complains that it can't detect the file type and does not load the file.

```
; XNC file generated by gerbonara
M48
METRIC
%
M30
```

AFAICT this is a specification-compliant empty XNC file, and I think gerbv should be able to detect it from the `M48` and `M30` commands.

ooxi, 2023-04-21:
Thank you for this report! Unfortunately I'm not too familiar with the XNC file format. Is that different from Gerber?

It would be great if you could provide a PR for the missing functionality. I would be more than happy to aid you in this endeavor.

eyal0, 2023-04-26:
Seems like XNC is a subset of Excellon that was invented to deal with the difficulty of getting the Excellon specification legally.  I think that if we just recognize these as Excellon files we should be fine.

https://www.ucamco.com/files/downloads/file_en/305/xnc-format-specification_en.pdf?1af7da772950b137447931f5470712f0

closed 2023-05-26 by ooxi

ooxi, 2023-05-26:
@jaseg: PR #170 by @eyal0 was merged as 8850f0b3f1813439f6437f04d861aeda2d2d85d0. Would be great, if you could give it a try!

jaseg, 2023-06-03:
Work for me, thanks!