gerbv/gerbv #169: Request for new features...

mikhailmihalkov, 2023-04-22:
Request for new features...
1 Open gerber files with a double mouse click when assigning gerbv to the default viewer (win7-11)
2 Make it possible to turn off or delete the aperture macro in an individual version of editing (replacing a complex form with a primitive and vice versa)
3 Add Interface Language Selection
4 Fix Errors in Fast Rendering Mode
one of the https://github.com/gerbv/gerbv/issues/167 or completely disable this mode
5 Make it possible to remember the program settings after restarting
6 Make a donation page for developers so that people can help the financial project

mikhailmihalkov, 2023-04-27:
Are there living people here?)))

ooxi, 2023-05-26:
@mikhailmihalkov thank you very much for your suggestions! While we are keen on improving gerbv/libgerbv, our resources are limited.

I am very keen on supporting you implementing features! My suggestion is picking a feature and we'll discuss implementation strategies :)

mikhailmihalkov, 2023-05-26:
I propose to start with a simple .... Include language packs in the assembly for csht as it is done in Linux ... The second thing you need to do is to make a link to the donation on the main page ...

ooxi, 2023-05-26:
@mikhailmihalkov that is a good pick! A common approach to select the language at runtime using a `gettext` based translation scheme is using [setlocale](https://docs.weblate.org/en/latest/devel/gettext.html#sample-program).

There might be some difficulties in changing strings after they have been loaded. Feel free to experiment! If you need help with anything gerbv specific feel free to ping me :-)