gerbv/gerbv #174: v2.9.6 Trailing zero imports wrong

htot, 2023-05-05:
In v2.9.6 (Ubuntu Lunar 23.04) I have:
```
M48
;Layer_Color=9474304
;FILE_FORMAT=2:5
INCH,LZ
;TYPE=PLATED
T1F00S00C0.00800
T2F00S00C0.01200
%
T01
X0060743Y0023895
X006281Y0025273
X0062909Y0022615
...
```
The (Altium generated) drill file is auto detected as None/Inch/5. Obviously 1st and 3rd drill are shown correctly, 2nd drill is placed near the origin (x = 0.06281).

Manually setting Trailing/Inch/2 all drills disappear entirely.

This is a regression as in 2.7.x and 2.8.x this worked correctly.

htot, 2023-05-05:
I don't know. It auto detects wrong in this case.

But worse is that after manual setting the correct setting, it doesn't show any drill holes at all.

htot, 2023-05-06:
I think I had the essence captured above, but here is the file.

[example.txt](https://github.com/gerbv/gerbv/files/11412184/example.txt)

eyal0, 2023-05-09:
Okay, let me see.  I'm trying a file that looks like this:

```
M48
;Layer_Color=9474304
;FILE_FORMAT=2:5
INCH,LZ
;TYPE=PLATED
T1F00S00C0.00800
T2F00S00C0.01200
%
T01
X0060743Y0023895
X006281Y0025273
X0062909Y0022615
M30
```

That second line in there is missing the final 0 in the X coordinate.  We are in leading zero mode so we expect that the file should behave as if that final zero is there, right?

Yup, looks wrong on my computer!

[image]

One of those holes is really far away from the others.  I'll will check older versions to see if this is a regression and when it happened!

eyal0, 2023-05-09:
Yup, as suspected, I broke the code in 944a4b3e8ba8e110d78e82606a914b8930a1e12b .  Damn.

I must say, I hate that leading zero and trailing zero is a feature of this language.  It's such a dumb idea!

Anyway, gotta fix it.

eyal0, 2023-05-09:
It's really confused by this:

```
;FILE_FORMAT=2:5
INCH,LZ
```

The `FILE_FORMAT` part implies that it will be 2 digits before the decimal and then 5 after.

And then the `LZ` implies that sometimes it *won't* be 2 and 5.  So what the hell is that file format doing there anyway?!?!

This thing is so annoying.  I'll take a look and see what I can do without breaking everything else.

htot, 2023-05-09:
It seems like we have multiple issues. I installed v2.9.2 and with auto detect on all goes well.

[image]

But when I manually set the format all drill disappear no matter what format I choose (you need to click Zoom to fit each time)

wdehoog, 2023-06-12:
@eyal0 To me it looks like there are two related problems in drill.c.

1. In drill_parse_header_is_metric_comment(...) where the FILE_FORMAT line is parsed the specified 'number of digits before' is not stored only 'number of digits before' is used.
2. When the FILE_FORMAT line is handled autod is set to 0 which causes the information of TZ or LZ to be ignored.

I guess both  'number of digits before' as well as 'number of digits before' need to be saved for later and after detecting LZ or TZ the correct one of them can be used (put into state->decimals).

eyal0, 2023-06-12:
That's what I determined, too. We need to store digits before, digits after, and whether it's LZ or TZ. Right now we only store two things.

I tried at it a little bit but stopped. The code is kind of a mess!