gerbv/gerbv #175: Soldermask subtract from silkscreen (Negative areas)

a-tawil, 2023-05-12:
Exported gebers using KiCad with turning (Soldermask subtract from silkscreen) on created some cross artifacts on the silkscreen gerber when viewed using gerbv.

Viewing with both KiCad gerber viewer and [Ucamco reference viewer](https://gerber-viewer.ucamco.com/) seems fine.
Refer to my issue on KiCad: [link](https://gitlab.com/kicad/code/kicad/-/issues/14721)

ooxi, 2023-05-26:
Thank you for reporting this @a-tawil! Would you mind attaching the gerber files causing the issue here? I looked at the linked KiCad issue but I did not see them. Maybe I overlooked?

a-tawil, 2023-05-30:
@ooxi , Sorry I mentioned on the KiCad issue that unfortunately I am not able to share the gerber or design files.
But I am pretty sure if you just try exporting gerbers from KiCad with the subtract option on you will get the same viewing results in gerbv.