gerbv/gerbv #180: Can not git clone on Windows

kitanokitsune, 2023-06-30:
When I try to git clone locally on Windows PC, an error occurs.
```
$ git clone https://github.com/gerbv/gerbv.git
Cloning into 'gerbv'...
remote: Enumerating objects: 11169, done.
remote: Counting objects: 100% (970/970), done.
remote: Compressing objects: 100% (408/408), done.
remote: Total 11169 (delta 556), reused 821 (delta 500), pack-reused 10199
Receiving objects: 100% (11169/11169), 20.75 MiB | 11.20 MiB/s, done.
Resolving deltas: 100% (7940/7940), done.
fatal: cannot create directory at '.mc/fedora:38': Invalid argument
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'
```
  
The reason is that the filepath (_e.g._ '.mc/fedora:38') contains a character that is not allowed on Windows.  

According to https://stackoverflow.com/questions/1976007/what-characters-are-forbidden-in-windows-and-linux-directory-names , the nine characters `<`, `>`, `:`, `"`, `/`, `\`, `|`, `?` and `*` are prohibited to be included in the name.

The following six path names apply to this case:
```
.mc/ubuntu:22.04
.mc/ubuntu:22.04/ubuntu:22.04.yaml
.mc/fedora:38
.mc/fedora:38/fedora:38.yaml
.mc/windows:amd64
.mc/windows:amd64/windows:amd64.yaml
```

Is it possible to fix this issue for Windows users?

ooxi, 2023-07-06:
Thanks for the report, we can fix that!

ooxi, 2023-07-09:
@kitanokitsune would be great if you could try a check #188

kitanokitsune, 2023-07-10:
Thank you very much!

This fix seems to work for my Windows environment!
```
$ git clone -b feature/windows-checkout https://github.com/ooxi/gerbv.git
Cloning into 'gerbv'...
remote: Enumerating objects: 11307, done.
remote: Counting objects: 100% (1108/1108), done.
remote: Compressing objects: 100% (473/473), done.
remote: Total 11307 (delta 633), reused 928 (delta 556), pack-reused 10199
Receiving objects: 100% (11307/11307), 20.79 MiB | 1.41 MiB/s, done.
Resolving deltas: 100% (8017/8017), done.
```

It seems that all files were successfully cloned without any errors.
```
$ ls -a
./                     CONTRIBUTORS           pixmaps/
../                    COPYING                po/
.git/                  desktop/               README
.github/               doc/                   README.md
.gitignore             dontdiff               README-git.txt
.mc/                   example/               README-release.txt
acinclude.m4           gerbv.github.io/       README-win32.txt
AUTHORS                HACKING                scheme/
autogen.sh*            icon-theme-installer*  share/
BUGS                   JOURNAL.txt            src/
build_release*         m4/                    Structs.txt
ChangeLog              Makefile.am            test/
ChangeLog_pre20080201  man/                   utils/
configure.ac           NEWS                   win32/
```

ooxi, 2023-07-10:
Patch merged

PR #188 (2023-07-10): Windows does not support colons in filenames
Fixes #180
commit b329acbd (2023-07-10): Windows does not support colons in filenames

Fixes #180
closed 2023-07-10 by ooxi