gerbv/gerbv #185: Zoom max level / GERBV_SCALE_MAX (and Windows build)

StephaneIFO, 2023-07-03:
Hi, 

In 2018, for a work colleague who wanted to make precision measurements, I recompiled under Windows a version of gerbv (2.6A) with an increased zoom level. 
At the time I modified the #define GERBV_SCALE_MAX from 3000 to 40000 in gerbv.h (https://github.com/gerbv/gerbv/blob/main/src/gerbv.h#L96C12-L96C12)

Now my colleague would like a more up-to-date version but with the same increased zoom limit.
Unfortunately, the Windows compilation doesn't seem to be up to date, so this ticket https://github.com/gerbv/gerbv/issues/19#issuecomment-931499319 seems to say that the "win32" directory is completely obsolete?

So, in order of simplicity for me: 
1) Can you change the max zoom limit in the official version?
2) If you can't, is there any up-to-date documentation for compiling a Windows binary?

Thanks, 
Stéphane

eyal0, 2023-07-04:
Are you able to use git bisect to find which is the first version in which you can't compile for windows?

eyal0, 2023-07-04:
I see successful builds for windows in CI. Can you follow the same procedure?

ooxi, 2023-07-06:
Maybe the Windows build is broken on Windows? We have received an issue that repository checkout doesn't work.

Nevertheless, I think there is no harm in increasing the zoom level in main.

@StephaneIFO I have prepared PR #186, would be nice if you could check out the Windows CI build of that

closed 2023-07-09 by ooxi

StephaneIFO, 2023-07-10:
@eyal0: It's complicated to reproduce the 2.6 build (old PC, Sourceforge SVN) and I'm not comfortable with "git bisect" and "automated build systems". My first goal isn't to recompile Gerbv but I'll follow this path if need be

@ooxi: I found this build https://github.com/gerbv/gerbv/actions/runs/5498639940 and the Windows version works perfectly! 

Attached are 2 max zoom images, you can see that the measurement line is coarser on 2.9.8 but it's already very good 👍 
The value of 40000 that I gave is the value I used at the time without really knowing the acceptable limit, but now is it possible to zoom in a little more? 

Thanks

[image]
[image]

ooxi, 2023-07-10:
@StephaneIFO sure we can increase the limit more. It would be great if you could suggest a reasonable value. If you clone this repository, you should be able to run the actions yourself so you have a windows binary after modifying the source online in GitHub.

That way we can iterate faster :-)

StephaneIFO, 2023-07-10:
As I said above, I'm not familiar with these automatic build systems, I've always compiled on my machine.
But if I make a fork and make changes in it, can I ask Github to compile for me?

eyal0, 2023-07-10:
@StephaneIFO We have automatic build of Pull Requests in github.  If you click the "fork" button on the main page, create your own fork, and then make a branch in there, you can make a pull request for the branch.  Pull requests are run through CI and you can see the output.  Here are some samples of CI runs:

https://github.com/gerbv/gerbv/actions

PR #186 (2023-07-09): Increase maximum zooming level
Fixes #185
commit aaa287cf (2023-07-09): Increase maximum zooming level

Fixes #185