gerbv/gerbv #189: "Apply to visible" translation breaks "Layer panel" display

StephaneIFO, 2023-07-10:
Hi, 

Now that we're using the latest version (thanks for the max zoom modification) it seems to me that a bug has appeared in the latest version v2.9.8. This bug is not present in v2.9.7.

I've attached an animated gif showing the sequence 
[image]

Step to reproduce (v2.9.8 Windows 10 22H2) : 
1) Display several layers
2) Right-click on a layer / Edit layer
3) Change translation values
4) Click on "Apply to visible"
=> "Layer panel" : random display of some layers, mouse/click seems correct
=> Next to each layer there's a red square instead of a "T" as in version 2.9.7 

The bug also seems to appear on the 2nd or 3rd time you do an "Apply to active".

Thanks.

StephaneIFO, 2023-07-10:
I'll confirm with my colleague tomorrow but there's no such bug on the "Version 2.9.7~742144" version.

The easiest way (as for max zoom) would be if I could do the build myself or via Github ;-)

eyal0, 2023-07-10:
https://github.com/gerbv/gerbv/actions

You can see builds here and each one has an artifact that you can unzip.  Inside, you'll find a program that you should be able to run.

StephaneIFO, 2023-07-11:
So we'll try these builds to see when the bug appears : 
https://github.com/gerbv/gerbv/actions/runs/5081645222
https://github.com/gerbv/gerbv/actions/runs/5090095611
https://github.com/gerbv/gerbv/actions/runs/5093712115

Thanks to both of you

StephaneIFO, 2023-07-11:
I'm not sure about the order of the commits/builds, I based myself on the time of the binary but it seems that the problem comes from "Update to Fedora 38 in CI".

gerbv_2023-05-25_9d5324_(Windows amd64) : ok
gerbv_2023-05-26_1364c2_(Windows amd64) : ok
gerbv_2023-05-26_8850f0_(Windows amd64) : ok
gerbv_2023-05-26_fc3073_(Windows amd64) :  bug  : Update to Fedora 38 in CI 
gerbv_2023-05-26_04c0ef_(Windows amd64) : ok  :  Updating CI to Fedora 38 is more cumbersome than expected #337
gerbv_2023-06-11_51d141_(Windows amd64) : bug : Release 2.9.8-rc.1
gerbv_2023-07-09_aaa287_(Windows amd64) : bug

[image]

ooxi, 2023-07-13:
@StephaneIFO thank you for the investigation! If upgrading the CI introduced a bug for you, it might have something to do with an updated GTK dependency for the Windows build.

In order to confirm this, I'll provide a new build with an old CI system. Might take a couple of days.

eyal0, 2023-07-13:
> I'm not sure about the order of the commits/builds, I based myself on the time of the binary but it seems that the problem comes from "Update to Fedora 38 in CI".

The CI doesn't necessarily run the code in the order of commits to `main` branch because PRs can sneak in.  Let me map what you've found to the history of main:

```
* b329acb - (gerbv/main, main) Windows does not support colons in filenames (3 days ago)
***BUG*** * aaa287c - Increase maximum zooming level (4 days ago)
* 356cea4 - focus-in-event callback return gboolean (4 days ago)
* dca2805 - Fix: Edit-layer is not editable (4 days ago)
* 2582ed0 - open .gvp (7 days ago)
* 929ca54 - Fix: crash Edit-file-format dialog (7 days ago)
* f45c994 - Japanese Language File (7 days ago)
* c604eff - (tag: v2.9.8) Release 2.9.8 (3 weeks ago)
***BUG*** * 51d1414 - (tag: v2.9.8-rc.1) Release 2.9.8-rc.1 (5 weeks ago)
***OK*** * 04c0ef7 - Updating CI to Fedora 38 is more cumbersome than expected (7 weeks ago)
* 08501bc - Include PR#177 in NEWS (7 weeks ago)
***BUG*** * fc30735 - Update to Fedora 38 in CI (7 weeks ago)
* c8bfe73 - Include PR#170 in NEWS (7 weeks ago)
***OK*** * 8850f0b - fix: Recognize empty drill files. (7 weeks ago)
* f605342 - Include PR#172 in NEWS (7 weeks ago)
***OK*** * 1364c29 - Bump APERTURE_PARAMETERS_MAX to 10006 (7 weeks ago)
* c559eac - Include PR#173 in NEWS (7 weeks ago)
***OK*** * 9d53242 - fix: Using floating point math for arcs. (7 weeks ago)
* b2d6fc1 - Bump version (7 weeks ago)
* 7421447 - (tag: v2.9.7) Release 2.9.7 (7 weeks ago)
* 7f57556 - (tag: v2.9.7-rc.1) Release 2.9.7-rc.1 (3 months ago)
* 44775d2 - Update BUGS file to point to issue tracker (3 months ago)
* 6b4f6ba - Bump version (6 months ago)
* 77211d1 - (tag: v2.9.6) Release 2.9.6 (6 months ago)
* 05a416f - (tag: v2.9.6-rc.1) Release 2.9.6-rc.1 (7 months ago)
* 17c6f68 - (meantaipan/main) Include PR#156 in NEWS (8 months ago)
* 944a4b3 - fix: Turn off autod after FILE_FORMAT comment. (8 months ago)
* c66083e - test: FILE_FORMAT togher with INCH statement. (8 months ago)
* 3ece126 - Bump version (8 months ago)
```

So this is kind of inconclusive because we see the bug at fc30735, then it goes away at 04c0ef7 and then returns at 51d1414 onward.  But it seems like 8850f0b and before should be "ok" and 51d1414 and after are problems.  Let's look at that diff: https://github.com/gerbv/gerbv/compare/8850f0b...51d1414

The difference is just the Fedora version!  Maybe something is wrong in Fedora?  Want to try a build that is `main` latest but without the Fedora change?  That will give us confidence that the problem is Fedora and not gerbv.

@StephaneIFO Please check this link in about 20 minutes and test on the artifacts that are created: https://github.com/gerbv/gerbv/actions/runs/5546714346  Thank you!

StephaneIFO, 2023-07-19:
It seems that the Fedora 36 build failed : 
[image]

ooxi, 2023-07-19:
In order to fix #180 we have renamed the CI files. This revert has been caught in a limbo.

@eyal0 remove build of Fedora and Ubuntu from GitHub Actions. This should likely fix the build.

StephaneIFO, 2023-08-28:
Hi,

Any news from the Fedora 36 build ?   https://github.com/gerbv/gerbv/pull/194

Thanks

eyal0, 2023-08-30:
I won't have time to work on this until perhaps next week.

eyal0, 2023-09-23:
Okay, I didn't get around to it.  I mean, it's hard to believe that the change in Fedora is what caused this issue.

I don't know enough about the changes to the Fedora build in order to work on this easily.  I won't be able to dig into this one.

StephaneIFO, 2023-09-25:
Hi,
The only clue at the moment is the switch from Fedora 36 to 38. Are there any lib version or compiler changes that could explain this bug?
A solution to confirm this hypothesis would be to restore a build based on Fedora 36.

I understand that you don't have much time to devote to this bug, and I'd love to help you, but my skills are limited!
I think PR #194 failed because you also need to revert the ':' to '_' changes from PR #180 in the ".github/workflows/ci.yaml" file.

```
from  run: npx --package mini-cross@0.15.2 mc --no-tty fedora_36 .mc/rebuild.sh
to    run: npx --package mini-cross@0.15.2 mc --no-tty fedora:36 .mc/rebuild.sh
```

Can you try this? 
Or if I make a fork can I make the modification myself and launch a build?

darkworks, 2024-06-20:
any update on this.

rampageservices, 2024-08-20:
This issue seems to have been running for a long time. May it be wise to bump to version 40 of Fedora and see if this issue persists?

Also, no one has mentioned if there are any STDOUT or STDERR during the process of testing this issue.

closed 2026-02-25 by spe-ciellt