gerbv/gerbv #195: Drill files get scaled incorrectly

henrygab, 2023-08-03:
May be related to #174 ??

* PCB Viewer in KiCad 6.0 shows everything aligned properly.
* JLCPCB.com online gerber viewer shows everything aligned properly.
* tracespace.io/view online gerber viewer shows everything aligned properly.

However, using GerbV will show the drills at the wrong location (wrong scale?).
Here's an ultra-tiny repro:

[GerbV_Bug.zip](https://github.com/gerbv/gerbv/files/12245790/GerbV_Bug.zip)

It appears by removing ALL the commented lines at the top of the drill files, the drills are then properly parsed / shown.

This is why I thought it might be related to #174.  (I'm opening a separate issue until it's determined to be the same.)

ooxi, 2023-08-18:
Thank you for the report! This seems to be related to #174 and provides a good test case for a fix!

While I currently do not have the capacity to provide a fix, I'm happy to provide guidance

henrygab, 2023-08-19:
> While I currently do not have the capacity to provide a fix, I'm happy to provide guidance

I have reviewed enough of the code to make a recommendation, and can likely provide a PR with fixes.  I'd appreciate your thoughts....

<hr/>
<details><summary><b>Recommendation (click to expand/collapse)</b></summary><p>

(All changes appear to reside in `src/drill.c`.)

* Define new struct `number_format_options_t`:
  * `uint8_t leading_digits` -- stores only the count of leading digits
  * `uint8_t trailing_digits` -- stores only the count of trailing digits
  * `gerbv_unit_t` -- stores the units (inches, mm, unspecified)
  * TBD ... if code actually requires `number_fmt_t` (e.g., to output a pre-defined format using short notation), then can also add that as a field of this new structure ...
* In existing `drill_state_t` structures:
  * TBD ... should existing `gerbv_unit_t unit` field be removed?
  * Replace three existing `number_fmt_t` fields with new `number_format_options_t` fields
* Adjust code accordingly (some comments already added above)
  * TBD ... add helper and internal consistency validation functions for `number_format_options_t`
* In addition, improve README file format requirements adherence / consistency 
  * convert all tabs into spaces
  * apply braces consistently
  * try to declare variables according to their needed scope (less mental load)
  * use more descriptive variable names where possible

</p>
</details><hr/>

The renaming / deletion of fields should cause a build break if anything is missed.  However, I'm really hoping the CI tests run lots of test cases.  If the build succeeds, but test cases fail, I will likely require assistance.

ooxi, 2023-08-19:
Those changes do look promising! I would like to split the changes into smaller chunks. Especially formatting might cause a huge diff and thus should be done once and backed up programmatically.

henrygab, 2023-08-19:
Yes, absolutely ... I am of like mind.  I just saw your [comments](https://github.com/gerbv/gerbv/pull/196#issuecomment-1684940913) in the **_[WIP]_** PR.

I will submit a first PR that is 100% programmatic formatting changes, powered by `clang-format`.

Three questions:
1. Do the existing CI test cases validate **_functionality_**? (or only build / code coverage?)
2. What is this project's preferred style:
    * LLVM
    * GNU
    * Google
    * Chromium
    * Microsoft
    * Mozilla
    * WebKit
3. Do you want the variable renaming / locality fixes as a separate PR, or is it enough for it to be a separate commit within the PR?

Thanks!

<hr/><details><summary>My current plan, for your thoughts</summary><p/>

1. `clang-format` + CI validation of same
2. Enable manual workflow initiation (`workflow_dispatch`)
--- PR for above formatting & CI improvements ---

1. adding comments on problem areas to address
2. changes that do not affect logical flow (e.g., localized & naming of variables)
3. individual refactorings (yes, `static` functions local to the file only ... so compiler can optimize away)
4. then, finally, the changes for #174

--- will ask for testing / confirmation here ---

</details><hr/>

henrygab, 2023-08-20:
@ooxi ... Please let me know if the clang reformat of PR #199 matches your expectations.  I don't want to spend time on the next steps until that is in `main` (for obvious refactoring reasons).

Thanks!

ooxi, 2023-08-21:
@henrygab I agree with not doing to many steps at once! We'll merge your changes and patch release :-)

henrygab, 2023-08-22:
I realize there is a long history for this project.  Accordingly, can you help me understand what minimum version of the C standard is required to be supported (across all supported platforms)?

e.g., am I restricted to C89 features?
e.g., can I safely rely upon the existence of C99 features (e.g., `stdint.h`)?
e.g., can I safely rely upon any C11 features (e.g., `_Static_assert(chk, msg)`)?

ooxi, 2023-08-23:
@henrygab there is no formal definition of minimum C standard. You can safely assume C11. Please refrain from using features which are not available in a compiler currently used by the CI unless there is a good reason from using them :-)

henrygab, 2023-08-25:
I do have two more questions, if you're still willing to help me grok more:

The HID Attribute thing ... on entry into `drill.c`, existing settings are used for default drill state values, and when done parsing, at least some attributes are stored into this HID attribute structure.  With some hand-waving, this appears to be a C implementation of a dictionary(*) storing variants, thus allowing generic-like programming, without the ease of using generic programming.  😄

1. Can you give me a high-level overview of what the attributes that `drill.c` adds to this structure do, **_outside of the `drill.c` file_**?

2. Are there existing functions somewhere that dump the HID_Attribute structures to the console (for logging / debugging purposes)?

BTW, I think I have code for parsing the fixed-number formats (including trailing / leading zero suppression) that is significantly simpler than what exists.

<details><summary>* (expand for less hand-waving)</summary><p/>

Not really a dictionary, more an array.  But each array entry stores its own field name, help text, value type, and (for enums) the stringification of the enum values.  I may find that this allows dumping state to a debugger, complete with common (generic style) code that converts these limited types to strings.

Thus, what I really mean to say is that it allows a programmatically opaque set of data to be stored, while allowing utility routines external to the data source to still be able to display that information for human consumption, even without understanding the programmatic meaning of the data.

</details>

henrygab, 2023-08-29:
Question on `HID_attributes` ...
What's the intent of these (at least as used in drill.c)?   

<details><summary>Chicken and Egg problem?</summary><p/>

One of the parameters to `parse_drillfile()` is a list of these.  However, the definitions are only within `drill.c`, so they're not populated until you call `parse_drillfile()`.   So how does any UI get to show these attributes (and thus allow users to edit them, in order to provide them to `parse_drillfile()`) in the first place?

I thought maybe this was intended to allow viewing / storing / editing of settings that are only defined internally to the drill.c file.  However, I'm having difficulty divining the rules that decide which of these values are updated in the `HID_attributes` during / after parsing a drill file.

For example, there are multiple test files where the value in `HA_digits` ends up with a different value than stored in the `drill_state_t`'s `decimals` field (e.g., when `number_fmt_t` is set to `FMT_USER`).

Thank goodness for the ~100 test cases.  I've pretty much just been adding debug information, testing various potential definitions of invariants, while trying not to cause any test failures.

Let's say ... it's clear that this code base has a **_long_** history.

</p>
</details>

henrygab, 2023-08-29:
Here's an easy question:

Are there constraints on the header comment-based number configuration format?
More specifically using example, **_are each of the following valid?_**

1. `;FILE_FORMAT=2:4` aka FMT_00_0000, imperial default (valid?)
1. `;FILE_FORMAT=3:3` aka FMT_000_000, metric (valid?)
1. `;FILE_FORMAT=3:2` aka FMT_000_00, metric (valid?)
1. `;FILE_FORMAT=4:2` aka FMT_0000_00, metric (valid?)
1. `;FILE_FORMAT=2:2` ... is it restricted to 5 or 6 digits total (historically true)? (invalid?)
1. `;FILE_FORMAT=1:5` ... if restricted to 5 or 6 digits, can it be non-standard? (valid?)
1. `;FILE_FORMAT=0:6` ... can it be without leading digits at all? (valid?)
1. `;FILE_FORMAT=6:0` ... can it be without decimal points at all?  (valid?)
1. `;FILE_FORMAT=9:7` ... can it be much larger number of digits? (???)
1. `;FILE_FORMAT=15:99` ... what limits should be enforced? (???)
1. `;FILE_FORMAT=0:0` ... for my own notes, handle this pathalogical edge case 🙈  (invalid)

eyal0, 2023-08-29:
I have no idea but we can follow the Poster Law

https://en.m.wikipedia.org/wiki/Robustness_principle

We should be as flexible as possible in what we accept. So I would like for us to support all the possibilities.

Your number_format_options_t above is missing the tz/lz flag. I suggest that you add an enum (not a Boolean) that specifies if we are in TZ mode or LZ mode. Or did you plan to store this elsewhere...?

As a review, tz means trailing zeros and if a number doesn't have enough digits to fill out the number of digits defined in the file format, it should add zeros to the end. Lz mode is leading zeros and the zeros should be added to the beginning. I think that I got those right and not reversed....

I think that this should be stored together in the same struct as the rest of the file format stuff. If the file has two file formats then we just use whichever one comes later.

henrygab, 2023-08-29:
Interesting.   I thought `TZ`/`LZ` setting should be included in the number format initially also.  However, after much review, I am currently thinking `TZ`/`LZ` are part of the machine's settings, and thus correctly reside in their **_current_** location (`image->format->omit_zeros`).

While the Gerbv code tracks which zeros to be **omitted** (trailing or leading), the excellon file format uses `TZ` and `LZ` to refer to which zeros to be **kept** when writing the file.  😞  Negating the logic level when transitioning between in-file and in-memory is ... an interesting design choice ... and made me wonder how anything was working for a while.  🤣 

For an example 1.5" measurement, with `FMT_00_0000`:

| File TZ/LZ |  Enum used | Encoded Value |
|:----------:|:-----------|---------------------:|
| none | GERBV_OMIT_ZEROS_UNSPECIFIED | 015000 |
| TZ   | GERBV_OMIT_ZEROS_LEADING     | 15000 |
| LZ   | GERBV_OMIT_ZEROS_TRAILING    | 015 |

Also, when I tried to integrate the LZ/TZ, I encountered numerous test failures.   There is some subtle interplay that I'm still trying to reason out.  Therefore, I'm slowly building back to such a larger change. 

As for the file format settings, I will explicitly document the limitation ... and there will be a limitation ... such that if that limit is exceeded, a clear error message will be generated.

eyal0, 2023-08-29:
I also gave this a try at one point and had about as much fun as you are! 😂 I don't have any wonderful advice for you, I'm sorry.

henrygab, 2023-08-30:
While the Gerbv code tracks which zeros to be **omitted** (trailing or leading), the excellon file format uses `TZ` and `LZ` to refer to which zeros to be **kept** when writing the file.  😞  Negating the logic level when transitioning between in-file and in-memory is ... an interesting design choice ... and made me wonder how anything was working for a while.  🤣 

Notes for an example 1.5" measurement, with `FMT_00_0000`:

| TZ/LZ |  Enum used                     | Encoded  | Read Method  |
|:-----:|:-------------------------------|-----------:|--------------|
| none  | `GERBV_OMIT_ZEROS_UNSPECIFIED` | `015000`   | value / 10^4 |
| TZ    | `GERBV_OMIT_ZEROS_LEADING`     | `15000`    | value / 10^4 |
| LZ    | `GERBV_OMIT_ZEROS_TRAILING`    | `015`      | See note ...  |

The first two cases simply scale the value read by `10^n`,
where `n` is the number of digits following the decimal.

The third case needs to know (in addition to the fact that it omits trailing zeros):
* How many digits are expected prior to the decimal?
* How many digits were read from the value?

Then, the scale is trivially computed as:
`scale = 10^(leading_digits - digits_read);`

Putting this all together, the logic is simply:

```
double scale =
    (image->format->omit_zeros == GERBV_OMIT_ZEROS_TRAILING) ?
    10.0 * (leading_digits - digits_read) :
    10.0 * trailing_digits;
```

However, as seen in the following test files, the selection of leading/trailing zero omission may occur after the `;FILE_FORMAT=` line:
* `Altium_file_format_inch.drl` -- has the header comment `;FILE_FORMAT=3:2` first
* `Altium_inch_file_format.drl` -- has the header directive `INCH,TZ` first

Thus, because the leading/trailing zero option may be set after selecting `FMT_USER`, it appears that supporting the ommision of trailing zeros (e.g., using `TZ`) requires **_both_** the digit count prior to the decimal **_and_** the digit count following the decimal.

TODO: Determine if these two Altium test cases are both properly parsed today.

henrygab, 2023-09-01:
I have a working fix.  :tada:

* Lots of validation / invariant checking code
* 100% pass rate of existing tests
* Fixes #195
* Adds a new value to HID_Attributes for drill.c ... I still don't know how these are used "in the wild".

So... now comes the question of taste / style / preferences for this depot.  I'd like to do the cleanup to help make this best match your desires.

1. **Do you want the validation and invariant checks included?**
    On the positive side, they are quite helpful in catching code errors,
    as well as catching inputs that fail to meet expectations.
    On the negative side, there is a risk that a valid input file exists
    with no corresponding test case, and that this causes an invariant
    to fail.   That said ... that's a good way to ensure the bug is more
    easily triaged and helps with root-cause-analysis. I lean towards
    **_keeping_** the invariant checks.
    
2. **Do you want the equivalent of a squash merge, or do you want
    a plurality of smaller commits?**
    Presume that any commits can be checked out, and it will not
    only compile, but will also pass all the automated tests.
    I lean towards a **_plurality_** of smaller, focused commits.

3. **Where a commit runs code in parallel, do you want me to squash it?**
    Especially during refactoring, I ran old code and new code, comparing
    results, to ensure no bugs crept in.  Somewhat noisy if left in, but also
    gives more depth to `git bisect`.  I lean towards **_removal_** of these.

<details><summary>Sample list of changes</summary><p/>

1. Add two new test cases to cover `INCH,LZ` + `;FILE_FORMAT=2:4` in one drill file.
1. Add `stdbool.h` (added in C99, so well before C11 target)
1. Re-ordered parameters of `drill_parse_coordinate()` for consistency
1. Removed unused parameter from `drill_state_t new_state()`
1. Removed unused parameter from `drill_add_drill_hole()`
1. Allow source attribute to be `const` for `drill_attribute_merge`
1. M70/M71 ... invariant check added for backup number format (when autodetecting file format)
1. `gerbv_HID_Attribute` - `name` and `help_text` fields marked `const`
1. Added various utility functions to convert enums to strings
1. Major rewrite / simplification of `read_double()`
1. Added many invariant checks for `state` & `image`; See code for details, but examples include:
    a. floating point values remain regular values (not INF, -INF, NAN, etc.)
    b. enumerations remain within valid range
    c. various integers remain within expected range
    d. when in data section, consistency of units and format type

</details>

## NOTE

There is still a need to add a couple test cases.   I also want to create a struct to wrap the format type, leading digits, and trailling digits (will make code cleaner and easier to understand).  So, while it's not quite ready for a PR, it's now a solved problem.   :tada:

henrygab, 2023-09-03:
Follow along as I compact the fixes into smaller, incremental improvements: https://github.com/henrygab/gerbv/pull/1

### Early review and feedback requested

Importantly, I've fixed the check for `clang-format` to use the checked-in formatting.  I could not get it to pass without this change.

I just added the final *required* commit to fix this issue.  However, I think **_there may be some edge cases_**:

1. when swapping between inch/metric via commands
2. when reading tools in the header

Both of those should be addressed by converting to a struct that holds all three items:  `{ number_format, leading digits, trailing digits }`

I also need to know if you want the additional invariant checks.

<details><summary>I squashed some commits already...</summary><p/>

I squashed the changed that showed **_equivalence_** of the old `read_double()` code's results vs. the new `read_double()` code's results.  Since it's floating point, the interpretation I took for "equivalence":

```C
double old_result = ...;
double new_result = ...;
double abs_difference = (old_result < new_result) ? new_result - old_result : old_result - new_result;
double allowed_difference = pow(10.0, -trailing_digits);
if (abs_difference > allowed_difference) {
```
</details>

henrygab, 2023-09-04:
## Review and decision required

Per the Excellon file specification, there is a maximum of 100 tools that can be defined in a drill file.  See https://github.com/henrygab/gerbv/blob/d3c32dfdbddd1111f10491891e5cf108f36df009/src/drill.c#L429-L434.

Accordingly, in my check of invariants, I had coded that tool number may never exceed 99.  However, I later discovered code that may allow for tools up to 9999.

#### Q1: Do you wish to allow tools beyond 99 to be parsed?**

#### Q2: If you do, how do you want to parse a line `T101`?

e.g.,  Is that tool `1` with compensation index `01`? _(complies with XNC specification)_
e.g.,  Is that tool `10` with compensation index `1`?
e.g.,  Is that tool `101` without any compensation index?

#### Q3: Similarly, how would you want to parse a line `T1007`?

e.g.,  Is that tool `10` with compensation index `07`? _(complies with XNC specification)_
e.g.,  Is that tool `100` with compensation index `7`?
e.g.,  Is that tool `1007` without any compensation index?

FYI ... all tests pass with tools limited to 99.

eyal0, 2023-09-23:
Um, I dunno!  I would usually prefer `gerbv` to be permissive, accepting even invalid code and trying to make it work with a warning printed.

However, in this case, there is ambiguity because having too many tools looks the same as the XNC compensation.

In practice, I think that having more than 100 tools is going to be really rare!  So I would expect that any file that specifies a big tool number is probably using a compensation index and not just having lots and lots of tools.

I guess that another thing that we could do it look at how many tools are defined.  If we see tool 101 defined but we also see that all tools 0...100 defined, well, maybe there really just is a lot of tools!

henrygab, 2023-09-27:
<details><summary>Archived: Why limit active tools to 99 ... reduce thread scrolling</summary><p/>

OK.  It would appear to be an unresolvable error to have more than 99 tools, at least because the specification provides no way to differentiate between the above situations.

For your awareness:

In PR#210, this limit of 99 tools is enforced.  Should the current tool ever exceed 99, `gerbv` will fail with a descriptive fatal error message, allowing for quickly determining the underlying problem with the file.  Note that this does not prevent **_defining_** more than 99 tools ... only attempts to try to **_use_** a tool greater than 99.

<details><summary>Details</summary><p/>

The invariants are checked in two spots:
1. after initialization and parsing of parameters
2. for each iteration of the main parsing loop

https://github.com/gerbv/gerbv/blob/aac8cddaa4175a4b7a005b9879e7254b0011844a/src/drill.c#L725-L732

`CHK_STATE` is just a macro that results in calls to `check_invariants()`.  `check_invariants()` defines the limit directly as the smaller of 99 tools, or the number of aperture slots.  (If aperture slots is ever reduced to less than 99, this prevents / detects memory corruption.)

https://github.com/gerbv/gerbv/blob/aac8cddaa4175a4b7a005b9879e7254b0011844a/src/drill.c#L425-L431

It later calls the helper function `check_min_max_int_value()` to validate that `state->current_tool` remains in this valid range.

https://github.com/gerbv/gerbv/blob/aac8cddaa4175a4b7a005b9879e7254b0011844a/src/drill.c#L459-L461

`check_min_max_int_value()` prints a descriptive fatal error message if this value ever falls outside this range:

https://github.com/gerbv/gerbv/blob/aac8cddaa4175a4b7a005b9879e7254b0011844a/src/drill.c#L405-L410

</details> 

Thus, even if this ambiguous situation were to ever occur, the current code would "safely" exit, and provides breadcrumbs to the comments on why `max_valid_tool` was defined as 99.   This is an acknowledgment of the uncertainty in my question (and your answer), by ensuring debuggability of the system if this unexpected condition should ever occur.

</details>

henrygab, 2023-12-01:
@eyal0 @ooxi 

## **_I intend to delete all my branches and related notes_**.

As I had previously noted [on September 21st](https://github.com/gerbv/gerbv/pull/210#issuecomment-1730833304), my memory has limits.  Given that multiple months have passed after having provided the full-featured fixes (with unit tests), even after refactoring the changes multiple times and being responsive to your feedback, I am left with few options.   Therefore, with some sadness, I am closing down all my (current and unreleased) efforts related to your project.

As a courtesy, I **_expect_** to defer this deletion for at least a week, although it may occur sooner.   Accordingly, should you wish to clone any of the branches, it is recommend that you do so as soon as possible.

Until next time...