gerbv/gerbv #198: BUILD HELP -- prerequisites are incomplete?

henrygab, 2023-08-19:
### Questions:

1. What package is referred to by `Cairo > 1.2.0`?
    There are multiple packages shown with `cairo` in their name by `apt list *cairo*`

2. `configure` notes this is required by `virtual:world`... but `apt list *vir*world*` doesn't show any matches.
    Which package is `virtual:world` referring to?

(above is run under KaliLinux, which is intentionally bare-bones)

<hr/><details><summary>Details (click to expand / hide)</summary><p/>

After running `./autogen.sh`, running `./configure` reports the following errors:
```
checking pkg-config is at least version 0.9.0... yes
./configure: line 21509: GLIB_GSETTINGS: command not found
checking for sin in -lm... yes
checking for dlopen in -ldl... yes
checking for cairo >= 1.2.0... no
configure: error: 
*** Cairo > 1.2.0 required but not found. ***
Please review the following errors:
Package 'cairo', required by 'virtual:world', not found
```

This is after fixing some other pre-requisites that appear to be missing from `README-git.txt`:

* [x] `autopoint`
    *  `sudo apt install autopoint`
    * on older distributions, may be part of the monolithic package: `apt list gettext`
* [x] `imagemagick`
    * `sudo apt install imagemagick`

</details>

ooxi, 2023-08-21:
We are using the following dependencies on [Fedora 38](https://github.com/gerbv/gerbv/blob/main/.mc/fedora_38/fedora_38.yaml#L14) and [Ubuntu 22.04](https://github.com/gerbv/gerbv/blob/main/.mc/ubuntu_22.04/ubuntu_22.04.yaml#L13).

The documentation with regards to dependencies could use an update, we are open to suggestions :-)

henrygab, 2023-08-21:
That was helpful, and Thank you!   These configurations for CI listed the necessary prerequisites, and I have now successfully built locally.

As a newcomer to the repo, it appeared that the file [`README-git.txt`](https://github.com/gerbv/gerbv/blob/76ec1da56e37de3e069426303aec1857e31dcd26/README-git.txt#L6-L35) was intended to inform folks of the dependencies.  However, it is clear that this listing is incomplete, and there was no obvious pointer from this existing documentation towards resolution.

Accordingly, please consider adding verbiage that points folks to these files, as the CI configuration already does the job of tracking the full set of dependencies needed (making this a low-effort, high-impact change).   e.g., something like....

For listings of dependencies used during GitHub automated builds, see [Fedora 38](https://github.com/gerbv/gerbv/blob/main/.mc/fedora_38/fedora_38.yaml) or [Ubuntu 22.04](https://github.com/gerbv/gerbv/blob/main/.mc/ubuntu_22.04/ubuntu_22.04.yaml).

Thank you for your consideration.  _I appreciate the work you folks are investing, and your responsiveness!_

ooxi, 2023-08-22:
You are correct and this is an artifact of the project's history. Since we rebooted development on GitHub, only `README.md` has been updated.

All other READMEs most likely refer to out of date information. While I was hesitant to remove them in the past, maybe they actually contain negative value by confusing newcomers?

henrygab, 2023-08-22:
The `INSTALL` file is good, but didn't give any hints as to these prerequisites.

I'm OK with removal of out-of-date documentation.  At the same time, it would be helpful for whichever documentation is kept to at least point folks to the existing prerequisites configurations.

As an example, as one who never used / heard about `Cairo2`, it was not clear which package was referred to by the error message, and `apt list *cair*` lists **well over 100 packages**!   Since the CI configuration already needs to track some of these prerequisites, that seems like a good place to point folks to.

Writing documentation that is clear and concise is ... not my strength.   Would you want me to leave this issue open?

ooxi, 2023-08-23:
@henrygab yes, let's keep this issue open. When I find time I'll create a PR removing some deprecated documentation and providing pointers for build in the README.md