gerbv/gerbv #233: rotate macro_line tiwce when export image to rs274x

chenshiqu, 2024-01-15:
case GERBV_APTYPE_MACRO_LINE22:
						/* Lower left line rectangle */

					/* Using LINE21 parameters array
					 * indexes for LINE21 and LINE22, as
					 * they are equal */
					if (trans->scaleX == trans->scaleY) {
						sam->parameter[LINE21_WIDTH] *=
								trans->scaleX;
						sam->parameter[LINE21_HEIGHT] *=
								trans->scaleX;

					} else if (fabs(sam->parameter[LINE21_ROTATION]) == 0
					|| fabs(sam->parameter[LINE21_ROTATION]) == 180) {
						sam->parameter[LINE21_WIDTH] *=
								trans->scaleX;
						sam->parameter[LINE21_HEIGHT] *=
								trans->scaleY;

					} else if (fabs(sam->parameter[LINE21_ROTATION]) == 90
					|| fabs(sam->parameter[LINE21_ROTATION]) == 270) {
						double t;
						t = sam->parameter[LINE21_WIDTH];
						sam->parameter[LINE21_WIDTH] =
							trans->scaleY *
							sam->parameter[
								LINE21_HEIGHT];
						sam->parameter[LINE21_HEIGHT] =
							trans->scaleX * t;
					} else {
						/* TODO: make outline macro */
						err_scale_line_macro++;
						break;
					}

					sam->parameter[LINE21_CENTER_X] *=
								trans->scaleX;
					sam->parameter[LINE21_CENTER_Y] *=
								trans->scaleY;

					sam->parameter[LINE21_ROTATION] +=
						RAD2DEG(trans->rotation);
                                        /************************ operation here is redundant  *******************************/
					gerbv_rotate_coord(
					        sam->parameter +LINE21_CENTER_X,
					 	sam->parameter +LINE21_CENTER_Y,
					 	trans->rotation);
                                       /*************************************************************/
					break;

eyal0, 2024-01-16:
Are you able to test it? Why do you say that it is redundant?

chenshiqu, 2024-01-22:
you can test this file. Use gerbv to load this file, rotate 180 degree, then export to rx-274x. Reload the exported file, you can see the issue.

eyal0, 2024-01-22:
Where's the file?

chenshiqu, 2024-01-24:
see the attachment.