gerbv/gerbv #238: Gerbv 2.10.0 or 2.10.1 Issues with PTH NCdrill locations

ejouellette, 2024-06-27:
I've submitted a 2 layer PCB design completed with Altium Designer 24.6.1.
Gerbers and NCDrill files attached in ZIP archive.
I understand that Gerbv doesn't yet support PTH slots, but it seems to have problems with the regular PTH locations.
They view fine in Gerbv 2.6.1a beta.

Thanks for looking into this!
Best,
Ed Ouellette
[Files.zip](https://github.com/user-attachments/files/16006558/Files.zip)

rampageservices, 2024-08-20:
Can you please outline what the problems were? Screenshots would help in diagnoses.

ejouellette, 2024-08-20:
Screenshot 2024-08-20 125442.jpg shows the Gerber + NCDrill layers all enabled for the project. Note the unexpected holes located outside the board area, on the left and lower sides.
Screenshot 2024-08-20 125510.jpg is the same Gerbers + RectHoles-Plated NCDrill layers, except Plated NCDrill layer has been disabled. Note that the erroneous holes outside the board area are now gone.
[image]
[image]

ejouellette, 2026-02-23:
> Thanks Ed!
> I will do a small clean-up and then push the fix, hopefully within a day
> or two.
> My plan is to use the drill file as a reference in the regression tests
> and the whole set of files will end up in the examples directory.

Sounds good, thanks.

Ed