gerbv/gerbv #245: Segfault on Fedora 41

PatrickFranken, 2024-11-13:
On startup gerbv causes a segmentation fault. This is the backtrace
```
Thread 1 "gerbv" received signal SIGSEGV, Segmentation fault.
0x00007ffff774eca9 in g_type_check_instance_is_fundamentally_a (type_instance=0x130000008f, fundamental_type=0x50) at ../gobject/gtype.c:4151
4151	  if (!type_instance || !type_instance->g_class)
(gdb) bt
#0  0x00007ffff774eca9 in g_type_check_instance_is_fundamentally_a (type_instance=0x130000008f, fundamental_type=0x50) at ../gobject/gtype.c:4151
#1  0x00007ffff7732fe7 in g_object_ref (_object=_object@entry=0x130000008f) at ../gobject/gobject.c:4240
#2  0x00007ffff7f3603e in IA__gdk_window_set_icon (window=0x63e130 [GdkWindow], icon_window=0x0, pixmap=0x0, mask=0x130000008f) at x11/gdkwindow-x11.c:3934
#3  IA__gdk_window_set_icon (window=0x63e130 [GdkWindow], icon_window=icon_window@entry=0x0, pixmap=0x0, mask=0x130000008f) at x11/gdkwindow-x11.c:3902
#4  0x00000000004142c2 in set_window_icon (this_window=0x515e50 [GtkWindow]) at interface.c:95
#5  interface_create_gui (req_width=req_width@entry=-1, req_height=req_height@entry=-1) at interface.c:1699
#6  0x0000000000404256 in main (argc=<optimized out>, argv=<optimized out>) at main.c:1075
(gdb) 
```

This is the system on which the issue occurs
```
# System Details Report
---

## Report details
- **Date generated:**                              2024-11-12 09:25:46

## Hardware Information:
- **Hardware Model:**                              Framework Laptop 13 _Intel Core Ultra Series 1_
- **Memory:**                                      32.0 GiB
- **Processor:**                                   Intel® Core™ Ultra 5 125H × 18
- **Graphics:**                                    Intel® Graphics (MTL)
- **Disk Capacity:**                               250.1 GB

## Software Information:
- **Firmware Version:**                            03.01
- **OS Name:**                                     Fedora Linux 41 (Workstation Edition)
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               47
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.11.4-301.fc41.x86_64

```

This patch obviously solved the issue
```
diff --git a/src/interface.c b/src/interface.c
index 8da9caf..a83471a 100644
--- a/src/interface.c
+++ b/src/interface.c
@@ -1696,7 +1696,7 @@ interface_create_gui(int req_width, int req_height) {
     callbacks_change_tool(NULL, (gpointer)0);
     rename_main_window("", mainWindow);
 
-    set_window_icon(mainWindow);
+    // set_window_icon(mainWindow);
     callbacks_update_layer_tree();
 
     /* Set GTK error log handler */
```

Root cause is probably the now disabled XPM image loader in new versions of gdk-pixbuf, see https://gitlab.gnome.org/GNOME/gdk-pixbuf/-/commit/e052a112075a19fb75f1f2ff3de4c82923de13f2

ion201, 2024-12-10:
I think you can close this issue. It is a packaging issue, not an issue with gerbv. I resolved this on my system by installing `gdk-pixbuf2-modules-extra`.

Reported downstream against fedora - https://bugzilla.redhat.com/show_bug.cgi?id=2331596

PatrickFranken, 2025-02-17:
This workaround works fine, will close the issue,

closed 2025-02-17 by PatrickFranken