gerbv/gerbv #248: gerbv does not respect operator precedence in aperture macros (this is explicitly required by the gerber spec)

metacollin, 2025-03-14:
Arithmetic operations are supported and part of the gerber spec within aperture macro primitives. 

For example, using the rect (`20`) primitive as an example, one would have:
```
20,1,$1+$1,$2,$3,$4,$5,$6x90*
```

which would multiply the 6th parameter passed to the macro by 90 and pass that as the rotation to the rect primitive.

From the gerber spec, section 4.5.4.2 (Arithmetic Expressions): 

> A parameter value can also be defined by an arithmetic expression consisting of integer and
> decimal constants, other variables, arithmetic operators and the brackets ‘(‘ and ‘)’. The
> **standard arithmetic precedence rules apply**. 

(emphasis mine).

However, if one does something like 
```
20,1,$1+$1,$2,$3,$4,$5,0*30+60
```
arithmetic precedence would apply the multiplication first, then add 60, giving a final value of 60. 

gerbv incorrectly calculates the value as 1860. Instead of respecting arithmetic precedence, it adds 60 to 0 and then performs the multiplication. 

gerbv also does not respect parentheses. The above issue should be easily worked around by doing something like:
```
20,1,$1+$1,$2,$3,$4,$5,(0*30)+60
```
but gerbv seems to ignore parentheses, as using them doesn't have any impact here.

spe-ciellt, 2025-03-20:
That is not true, see https://github.com/gerbv/gerbv/blob/main/src/amacro.c#L90 and forward.
It is a simplified implementation of the Djikstra "Shunting Yard" implementation. Simplified for example as it does not need to support parenthesis. The precedence algorithm works on the "assembler" that is generated by the parser.

The example you have given above, is that "home made" or is it actually generated from a CAD program?
If that is an actually allowed aperture macro, please check what aperture macro assembler is generated from the parser and make sure that is correctly generated.

metacollin, 2025-04-23:
@spe-ciellt Yes, I see that. But parenthesis or no, my point is that there is a bug somewhere in that code that is causing order of operations to **not** be respected. Try it yourself if you don't believe me. No parentheses needed. Those were just an attempt to work around (by explicitly forcing) order of operations to behave correctly. But you're right, they shouldn't be needed. 

Yes, it is from a real gerber generated by a CAD program. I can't provide the file unfortunately without the customer's permission which I don't have but I can provide screen shots that clearly show that gerbv incorrectly renders this gerber while other programs do not:

Paste layer, rendered in Cuprum:

[image]

Tracespace.io:

[image]

KiCad's gerber viewer:

[image]

And finally, gerbv (obviously broken):

[image]

The parentheses were indeed added by hand, but this was simply an attempt to work around gerbv's incorrect application of arithmetic precedence (aka order of operations), but that unfortunately also didn't work, rendering it impossible to workaround in gerbv. 

The lines causing these problems are all something along the lines of:

`20,1,$1+$1,$2,$3,$4,$5,0X25.4+90*`

This is a real file from a real customer. It's weird and not actually needed, but CAD problems are notoriously stupid when it comes to the sort of output they sometimes generate, especially for things like gerbers. 

In this case, the rotation (the final argument) should be 90 degrees, as 0x25.4+90 = 90 degrees. However, I manually measured the angle of the odd pad rotation gerbv displays and it is 36 degrees off from the correct 90 degree angle (vertical or horizontal). 

25.4*90 mod 90 = 36

The only way rotations would end up like that is if gerbv thinks 0X25.4+90 = 2286, which is incorrect. 

0x25.4+90 does not equal 2286, but gerbv thinks it does. **This is wrong, full stop**. 

It should definitely not be marked as won't fix, it's a bug and violates the actual format specification of gerbers besides.

closed 2025-05-24 by metacollin