gerbv/gerbv #249: Unknown RS-274X extension found

ldoolitt, 2025-03-31:
Recent versions of KiCad (>=8?) have started creating Gerber files that have RS-274X extensions %TA% %TD% %TF% %TO%.
When loaded by gerbv, the rendering is still perfectly fine, but it fills the console and the Messages pane with complaints like
```
Unknown RS-274X extension found %TF% at line 3 in file "/redacted/test1.gbr"
```

I found a pretty small example that I attach here: [zest-Edge_Cuts.txt](https://github.com/user-attachments/files/19542537/zest-Edge_Cuts.txt)

The only mention of this issue I could find online was in an out-of-the-way place:
X2 extensions should be ignored, by react66, Aug 23, 2021
https://github.com/alyoshin/gerbv-dxf-builded/issues/1
Note the date: over four years ago! Maybe this issue will be better at reaching the right people.
In that issue, react66 included the helpful comment:
> Gerber X2 has adds four new commands (TF, TA, TO and TD) to attach attributes to a Gerber file. All pre-existing commands remain unchanged.

I can confirm the effect on the Debian-shipped gerbv versions 2.9.6 and 2.10.0.
I haven't yet built gerbv from github source to find out if it silently got improved.

ldoolitt, 2025-04-01:
I can confirm this effect is still present in gerbv built from sources, github main commit 7c1553da, Dec 14, 2024.

tingox, 2025-11-20:
Still present in Version 2.10-1-dev~180c67 (win10) downloaded from the web page.

eyal0, 2025-11-20:
Should those just be ignored by gerbv?