gerbv/gerbv #255: fails with gcc15

pgajdos, 2025-06-10:
This fixes the build for me:
```
diff -upr gerbv-2.10.0.orig/src/attribute.c gerbv-2.10.0/src/attribute.c
--- gerbv-2.10.0.orig/src/attribute.c	2025-06-10 10:14:28.144410089 +0000
+++ gerbv-2.10.0/src/attribute.c	2025-06-10 10:16:19.807030302 +0000
@@ -193,7 +193,7 @@ static void
 ghid_spin_button (GtkWidget * box, GtkWidget ** spin_button, gfloat value,
 		  gfloat low, gfloat high, gfloat step0, gfloat step1,
 		  gint digits, gint width,
-		  void (*cb_func) (), gpointer data, gboolean right_align,
+		  void (*cb_func) (GtkWidget *, void *), gpointer data, gboolean right_align,
 		  gchar * string)
 {
   GtkWidget *hbox = NULL, *label, *spin_but;
@@ -248,7 +248,7 @@ ghid_check_button_connected (GtkWidget *
 			     gboolean expand,
 			     gboolean fill,
 			     gint pad,
-			     void (*cb_func) (),
+			     void (*cb_func) (GtkToggleButton *, gboolean *),
 			     gpointer data, gchar * string)
 {
   GtkWidget *b;
diff -upr gerbv-2.10.0.orig/src/callbacks.c gerbv-2.10.0/src/callbacks.c
--- gerbv-2.10.0.orig/src/callbacks.c	2025-06-10 10:14:28.144410089 +0000
+++ gerbv-2.10.0/src/callbacks.c	2025-06-10 10:39:44.515373183 +0000
@@ -1066,7 +1066,7 @@ callbacks_toggle_layer_visibility_activa
 	if (screenRenderInfo.renderType <= GERBV_RENDER_TYPE_GDK_XOR) {
 		render_refresh_rendered_image_on_screen ();
 	} else {
-		render_recreate_composite_surface (screen.drawing_area);
+		render_recreate_composite_surface ();
 		callbacks_force_expose_event_for_screen ();
 	}
 }
@@ -2185,7 +2185,7 @@ callbacks_layer_tree_visibility_toggled
 	if (screenRenderInfo.renderType <= GERBV_RENDER_TYPE_GDK_XOR) {
 		render_refresh_rendered_image_on_screen ();
 	} else {
-		render_recreate_composite_surface (screen.drawing_area);
+		render_recreate_composite_surface ();
 		callbacks_force_expose_event_for_screen ();
 	}
 }
@@ -2437,7 +2437,7 @@ callbacks_remove_layer_button_clicked (G
 		if (screenRenderInfo.renderType <= GERBV_RENDER_TYPE_GDK_XOR) {
 			render_refresh_rendered_image_on_screen ();
 		} else {
-			render_recreate_composite_surface (screen.drawing_area);
+			render_recreate_composite_surface ();
 			callbacks_force_expose_event_for_screen ();
 		}
 	}
@@ -2468,7 +2468,7 @@ callbacks_move_layer_down_button_clicked
 			render_refresh_rendered_image_on_screen ();
 		}
 		else {
-			render_recreate_composite_surface (screen.drawing_area);
+			render_recreate_composite_surface ();
 			callbacks_force_expose_event_for_screen ();
 		}
 	}
@@ -2497,7 +2497,7 @@ callbacks_move_layer_up_button_clicked
 			render_refresh_rendered_image_on_screen();
 		}
 		else {
-			render_recreate_composite_surface (screen.drawing_area);
+			render_recreate_composite_surface ();
 			callbacks_force_expose_event_for_screen ();
 		}
 	}
@@ -2525,7 +2525,7 @@ void callbacks_layer_tree_row_inserted (
 				render_refresh_rendered_image_on_screen();
 			}
 			else {
-				render_recreate_composite_surface (screen.drawing_area);
+				render_recreate_composite_surface ();
 				callbacks_force_expose_event_for_screen ();
 			}
 			/* select the new line */

```

eyal0, 2025-06-10:
Can you create a PR out of it so that the CI will run?

closed 2025-06-11 by eyal0

pgajdos, 2025-06-12:
@eyal0 sorry, something to finish?

Thanks for making the commit.

eyal0, 2025-06-12:
I think that I've finished the task and it all works now.  Your changes are in the latest code now.

PR #257 (2025-06-11): fix: Use correct API for compiling with gcc15.
This fixes #255
commit 180c677c (2025-06-11): fix: Use correct API for compiling with gcc15.

This fixes #255